sipeed/MaixPy-v1 #281: ussl library is not included in master

valenzmanu, 2020-07-22:
This issue comes from https://github.com/sipeed/MaixPy/issues/105

Hello @junhuanchen, I tried today the last version in master but it does not include ussl:

```
>>> help('modules')
help('modules')
KPU               hashlib           pye_mp            uio
Maix              heapq             random            ujson
__main__          image             re                ulab
_boot             json              sensor            uos
_thread           lcd               socket            urandom
_webrepl          lodepng           struct            ure
array             lvgl              sys               usocket
audio             lvgl_helper       time              ustruct
binascii          machine           touchscreen       utime
board             math              ubinascii         utimeq
builtins          math              ucollections      uzlib
cmath             micropython       ucryptolib        video
collections       modules           uctypes           zlib
errno             nes               uerrno
fpioa_manager     network           uhashlib
gc                os                uheapq
Plus any modules on the filesystem
```

After that, I tried using the version in https://github.com/sipeed/MaixPy/tree/develop (which does have ussl) but the camera is not working. I'm using a Maixduino.

junhuanchen, 2020-07-22:
but the camera is not working. why ?

valenzmanu, 2020-07-23:
I'm sorry I wasn't clear in my previous comment. 

Here are 2 pictures of my project using [this version of MaixPy](https://dl.sipeed.com/MAIX/MaixPy/release/master/maixpy_v0.5.0_98_g7ec09ea):
[image]
[image]

And here is the exact same code using the [develop branch](https://github.com/sipeed/MaixPy/tree/develop):
[image]
[image]

valenzmanu, 2020-07-23:
I'm not sure how or where to debug what happened since the only thing I changed was the MaixPy version.

junhuanchen, 2020-07-24:
set sensor rgb565 color change to yuv422

valenzmanu, 2020-07-24:
Thanks, it worked.

closed 2020-07-27 by junhuanchen