sipeed/MaixPy-v1 #284: GPIO issues - it doesn't work with SR04

pritamghanghas, 2020-08-02:
Hi I have a couple of SR04, I don't have access to the single pin module that the maixpy ultrasonic module seems to be programmed for. So I wrote a small python driver for it.

The problem is it doesn't work on maixpy while it works flawlessly on esp32 with micropython. Ofcourse i have to change pin number but other that that, they code works. 

Here is my pin setup.

    fm.register(board_info.LED_B, fm.fpioa.GPIO0, force=True) # 12 - blue led
    fm.register(board_info.LED_G, fm.fpioa.GPIO1, force=True) # 13 - green led
    fm.register(board_info.LED_R, fm.fpioa.GPIO2, force=True) # 14 - red led

    #buzzer pin setup
    fm.register(board_info.PIN9, fm.fpioa.GPIO7, force=True) # 8 - buzzer

    # back panel gpio setup
    fm.register(board_info.MIC_ARRAY_DATA3, fm.fpioa.GPIOHS0, force=True) # 20 - back panel 1
    fm.register(board_info.MIC_ARRAY_DATA2, fm.fpioa.GPIOHS1, force=True) # 21 - back panel 2
    fm.register(board_info.MIC_ARRAY_DATA1, fm.fpioa.GPIOHS2, force=True) # 22 - back panel 3

    # pump and laser control
    fm.register(board_info.MIC_ARRAY_DATA0, fm.fpioa.GPIOHS3, force=True) # 23 - solenoid control
    fm.register(board_info.MIC_ARRAY_LED, fm.fpioa.GPIOHS4, force=True) # 24 - laser led control

    # gesture sonic pins
    fm.register(board_info.PIN15, fm.fpioa.GPIO3, force=True) # 15 - trig for gesture
    fm.register(board_info.PIN17, fm.fpioa.GPIO4, force=True) # 17 - echo for gesture
    
    #temp sonic pins
    fm.register(board_info.MIC_ARRAY_BCK, fm.fpioa.GPIO5, force=True) # 18 - trig temp
    fm.register(board_info.MIC_ARRAY_WS, fm.fpioa.GPIO6, force=True) # 19 - echo temp

Neutree, 2020-08-03:
what problem with ultrasonic lib?
https://github.com/sipeed/MaixPy_scripts/tree/master/modules/grove/ultrasonic

pritamghanghas, 2020-08-05:
I don't have a sessor that can work with one pin. The library API seems to take only 1 pin.

Neutree, 2020-08-06:
it's better to code with C for this sensor, we not test on two pins sonic sensor, you can modify code here:
https://github.com/sipeed/MaixPy/blob/7caf24572f54db97f19df9e08e0b8b0ef6ebb15e/components/micropython/port/src/modules/ultrasonic/src/ultrasonic.c#L15-L72

now it only use one pin, first set pin to output mode, triger sensor, and immediately change to input mode, then wait for ack, you can add one pin here replace one pin

pritamghanghas, 2020-08-09:
It took a long time to find a firmware version where everything works. For the time being I would like to stay with MaixPy. Here is code that should have worked. https://github.com/pritamghanghas/sipeed_playground/blob/master/hcsr04.py

Similar code with different pin assignments and intialization worked with ESP12E microPython environment without any issues.

junhuanchen, 2020-11-13:
Now I2C add soft-ware i2c.

your can try this.

https://github.com/sipeed/MaixPy_scripts/blob/master/hardware/demo_i2c.py