sipeed/MaixPy-v1 #285: Feature request: support for UXGA JPEG saving, add AVI frame, base resolution for windowing

easy-and-simple, 2020-08-03:
Hello Guys,
Would you add UXGA support as base resolution for windowing mode, and also for taking JPEG images that then to be saved as JPEG image file or added as frames in AVI? 
OV2640 can make JPEG images so main problem here is ability to switch OV2640 between compressed JPEG mode and uncompressed RGB mode

junhuanchen, 2020-08-04:
I don't know the implementation details, but it might work.

Can you describe the process of implementation？

Neutree, 2020-08-06:
* SXVGA support is a feature request, it's no implement now, same as #243 
* K210 can not use JPEG mode of sensor, for hardware limitation, just can use RGB565 or YUV mode

closed 2020-08-06 by Neutree

AhmadPanogari, 2020-08-19:
Any update for this Issue?

I also need that image size when taking pictures 

and is the OV5640 already supported?

ZoneMR, 2023-02-21:
> * K210 can not use JPEG mode of sensor, for hardware limitation, just can use RGB565 or YUV mode

What is the specific hardware limitation which would prevent using the JPEG output from the OV2640?