sipeed/MaixPy-v1 #288: EIO error while using the Adafruit_CircuitPython_SSD1306

CharlieYu4994, 2020-08-12:
This following code doesn't work

    class SSD1306(_SSD1306):
        def __init__(self, i2c, addr, height, width):
            self.buffer = bytearray(((height // 8) * width) + 1)
            self.buffer[0] = 0x40
            self.temp = bytearray(2)
            self.i2c = i2c
            self.addr = addr
            super().__init__(memoryview(self.buffer)[1:] ,width, height)
    
        def write_cmd(self, cmd):
            self.temp[0] = 0x80
            self.temp[1] = cmd
            self.i2c.writeto(self.addr, self.temp)
    
        def write_framebuf(self):
            self.i2c.writeto(self.addr, self.buffer)

But this following code works

    class SSD1306(_SSD1306):
        def __init__(self, i2c, addr, height, width):
            self.buffer = bytearray((height // 8) * width)
            self.temp = bytearray(2)
            self.i2c = i2c
            self.addr = addr
            super().__init__(memoryview(self.buffer) ,width, height)

        def write_cmd(self, cmd):
            self.temp[0] = 0x80
            self.temp[1] = cmd
            self.i2c.writeto(self.addr, self.temp)

        def write_framebuf(self):
            for data in self.buffer:
                self.i2c.writeto_mem(self.addr, 0x40, data)

It's easy to see that the write_cmd method is the same and works properly

I am not good at English, please forgive me.

Neutree, 2020-08-12:
没理解你什么意思, 是 `write_cmd` 函数有问题? 但是上下两份代码的函数看起来都是一样的, 反而看起来更像想要说`writeto_mem`函数参数不支持bytes对象,你可以用中文,更详细地描述

CharlieYu4994, 2020-08-12:
> 没理解你什么意思, 是 `write_cmd` 函数有问题? 但是上下两份代码的函数看起来都是一样的, 反而看起来更像想要说`writeto_mem`函数参数不支持bytes对象,你可以用中文,更详细地描述

是在 write_framebuf 中用 writeto 的时候传入一个 bytearray 类型的数据会报 EIO 的错，但是在 write_cmd 中使用 writeto 不报错。

CharlieYu4994, 2020-08-12:
在第一份代码中我尝试用 `writeto` 把整个「buffer」一次性发送出去，但是会报「EIO」的错，之后我摸索出来，用 `writeto_mem` 一个个把数据发出去就不会有问题，但是在 `write_cmd` 中使用 `writeto` 就没有任何问题，我就觉得很奇怪

Neutree, 2020-08-12:
https://github.com/sipeed/MaixPy_scripts/blob/master/modules/OLED/ssd1306/ssd1306k.py , 这里 @CaptainJackey 也是对寄存器 0x40进行写数据,而不是直接写数据, 是硬件差异?

CharlieYu4994, 2020-08-12:
应该是我的问题，貌似 adafruit 的那个 ssd1306 不太一样，它在 buffer 的最开始发送了一次 0x40 的地址后，就再也没发送过寄存器地址了，https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/master/adafruit_ssd1306.py，就感觉挺奇怪的，不管了，反正第二份代码能用

closed 2020-08-12 by CharlieYu4994

CharlieYu4994, 2020-08-12:
@Neutree 我又试了一下，第一份代码并不是完全不能用，它在发送了 865 个 byte 后报了 EIO 的错 
[image]
 我先在显存里放了点东西，把代码换成第一种后擦除显存，发现擦除了一部分，根据显示，停下的地方是第七页的第96个byte，算上一开始的指令，一共865个byte，整个显存大小是 1024byte

CharlieYu4994, 2020-08-12:
然后一开始那个 0x40 是「controlbyte 」
[image]  `0x40`的话，看数据手册貌似是没问题的 Continuation bit 确实置零了，代表后面的全是数据没有指令