sipeed/MaixPy-v1 #289: Failed to load kmodel from onnx

yub1ng, 2020-08-13:
If I load a kmodel which compiled by `ncc compile onnx` from sd card, the machine will disconnect with maixpy IDE after a few seconds .

[image]

[yolo_mini.kmodel.zip](https://github.com/sipeed/MaixPy/files/5066565/yolo_mini.kmodel.zip)

Neutree, 2020-08-13:
run in termiter instead of IDE,  and get error output, by the way which version of firmware you use

yub1ng, 2020-08-13:
> run in terminal instead of IDE, and get error output, by the way which version of firmware you use

Thanks for your reply. 

My firmware version is maixpy_v0.5.0_119_ge7ab30a.

And here is the output in the terminal:

```sh
>>> task = kpu.load('/sd/yolo_mini.kmodel')
kmodel, version:1263354956, arch:2
core dump: fault load
Cause 0x0000000000000005, EPC 0x00000000800cf52a
reg[00](zero ) = 0x000000008011e6f0, reg[01](ra   ) = 0x00000000800cf134
reg[02](sp   ) = 0x00000000802f3658, reg[03](gp   ) = 0x00000000801f8870
reg[04](tp   ) = 0x0000000000000000, reg[05](t0   ) = 0x0000001000000000
reg[06](t1   ) = 0x0000000000000100, reg[07](t2   ) = 0x000000000000007d
reg[08](s0/fp) = 0x00000000803fdd20, reg[09](s1   ) = 0x0000000000000000
reg[10](a0   ) = 0x00000000803fdd30, reg[11](a1   ) = 0xffffffffbf689a62
reg[12](a2   ) = 0x00000000000f0411, reg[13](a3   ) = 0x0000000000002000
reg[14](a4   ) = 0x0000000000002000, reg[15](a5   ) = 0x0000000000000401
reg[16](a6   ) = 0x0000000000000002, reg[17](a7   ) = 0x0000000000000009
reg[18](s2   ) = 0x00000000803fdd30, reg[19](s3   ) = 0x00000000803fdc00
reg[20](s4   ) = 0x0000000000000000, reg[21](s5   ) = 0x0000000000000000
reg[22](s6   ) = 0x000000008030dcc8, reg[23](s7   ) = 0x000000008030dda0
reg[24](s8   ) = 0x00000000802f3954, reg[25](s9   ) = 0x0000000000000000
reg[26](s10  ) = 0x000000008011e6f0, reg[27](s11  ) = 0x000000008011e6e8
reg[28](t3   ) = 0x0000000000000020, reg[29](t4   ) = 0x0000000000000002
reg[30](t5   ) = 0x0000000000000004, reg[31](t6   ) = 0x0000000000000000
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c140()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c150()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c160()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c188()
freg[10](fa0 ) = 0x39343533409745d1(), freg[-1610612736]() = 0x000000008011c1b8()
freg[12](fa2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c1e0()
freg[14](fa4 ) = 0x000000004ba7d8c0(), freg[1073741824]() = 0x000000008011c1f0()
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c200()
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c228()
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c238()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c248()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c258()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x000000008011c268()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c290()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x000000008011c2a0()
W (52576465911) SYSCALL: sys_exit called by core 0 with 0x539
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c130()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c140()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c150()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c160()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c188()
freg[10](fa0 ) = 0x39343533409745d1(), freg[-1610612736]() = 0x000000008011c1b8()
freg[12](fa2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c1e0()
freg[14](fa4 ) = 0x000000004ba7d8c0(), freg[1073741824]() = 0x000000008011c1f0()
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c200()
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c228()
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c238()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c248()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c258()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x000000008011c268()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c290()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x000000008011c2a0()
W (52576465911) SYSCALL: sys_exit called by core 0 with 0x539
reg[30](t5   ) = 0x0000000000000004, reg[31](t6   ) = 0x0000000000000000
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c130()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c140()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c150()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c160()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c188()
freg[10](fa0 ) = 0x39343533409745d1(), freg[-1610612736]() = 0x000000008011c1b8()
freg[12](fa2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c1e0()
[... 26 lines cut ...]
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c290()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x000000008011c2a0()
W (52576465911) SYSCALL: sys_exit called by core 0 with 0x539
reg[24](s8   ) = 0x00000000802f3954, reg[25](s9   ) = 0x0000000000000000
reg[26](s10  ) = 0x000000008011e6f0, reg[27](s11  ) = 0x000000008011e6e8
reg[28](t3   ) = 0x0000000000000020, reg[29](t4   ) = 0x0000000000000002
reg[30](t5   ) = 0x0000000000000004, reg[31](t6   ) = 0x0000000000000000
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c130()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c140()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c150()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c160()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c188()
[... cut ...]
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c238()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c248()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c258()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x000000008011c268()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x000000008011c290()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x000000008011c2a0()
W (52576465911) SYSCALL: sys_exit called by core 0 with 0x539
```

Neutree, 2020-08-13:
It's a bug when read from SD card, I'll fix it right now

yub1ng, 2020-08-13:
> It's a bug when read from SD card, I'll fix it right now

THX!!!!!!!

Neutree, 2020-08-13:
I upload update just right now, wait for some moment firmware can be generated

Neutree, 2020-08-13:
https://dl.sipeed.com/MAIX/MaixPy/release/master/maixpy_v0.5.0_120_g384ea9a

yub1ng, 2020-08-13:
> https://dl.sipeed.com/MAIX/MaixPy/release/master/maixpy_v0.5.0_120_g384ea9a

Thanks, but after I write the maixpy_v0.5.0_120_g384ea9a, there still occur some error.
[image]

Neutree, 2020-08-13:
not support ops https://github.com/sipeed/MaixPy/blob/384ea9a6eacb90e1971a6db0775d076391a74501/components/kendryte_sdk/include/sipeed_kpu.h#L21

yub1ng, 2020-08-13:
> not support ops
> 
> https://github.com/sipeed/MaixPy/blob/384ea9a6eacb90e1971a6db0775d076391a74501/components/kendryte_sdk/include/sipeed_kpu.h#L21

Thanks, but I don't know which ops is not supported, can you figure out?
```python
"""
@author Yubing
@date 2020/7/28
"""
from torch import nn

def same_padding(shape, kernel_size, stride):
    return ((shape - 1) * stride + kernel_size - shape) // 2

class DBL(nn.Module):
    def __init__(self, in_channels, out_channels, kernel_size):
        super(DBL, self).__init__()
        padding = same_padding(in_channels, kernel_size, stride=1)
        self.net = nn.Sequential(
            nn.Conv2d(in_channels, out_channels, kernel_size=kernel_size, stride=1, padding=padding),
            # nn.BatchNorm2d(out_channels),
            nn.LeakyReLU(0.1)
        )

    def forward(self, X):
        return self.net(X)

class MDBL(nn.Module):
    """
    MaxPool + DBL
    """
    def __init__(self, in_channels, out_channels, pool_stride):
        super(MDBL, self).__init__()
        self.net = nn.Sequential(
            nn.MaxPool2d(kernel_size=2, stride=pool_stride),
            DBL(in_channels, out_channels, kernel_size=3)
        )

    def forward(self, X):
        return self.net(X)

class YOLOMini(nn.Module):
    def __init__(self, num_classes, num_anchors):
        super(YOLOMini, self).__init__()
        self.out_channels = num_anchors * (5 + num_classes)
        self.backbone = nn.Sequential(
            DBL(3, 16, 3),
            MDBL(16, 32, 2),
            MDBL(32, 64, 2),
            MDBL(64, 128, 2),
            MDBL(128, 256, 2),
        )
        self.output = DBL(256, self.out_channels, 1)

    def forward(self, X):
        """
        :param X: n, 3, 240, 320
        :return: n, out_channels, 15, 20
        """
        X = self.backbone(X)
        return self.output(X)

```

yub1ng, 2020-08-13:
> not support ops
> 
> https://github.com/sipeed/MaixPy/blob/384ea9a6eacb90e1971a6db0775d076391a74501/components/kendryte_sdk/include/sipeed_kpu.h#L21

What if the model is right, but is the ncc problem or the maixpy problem. ONNX has many bugs in ncc...

Neutree, 2020-08-14:
maybe It's better convert to tflite first, then convert by nncase

yub1ng, 2020-08-14:
> maybe It's better convert to tflite first, then convert by nncase

Thanks, It may be a good short-term choice but cant be a long-term solution.

yub1ng, 2020-08-14:
> maybe It's better convert to tflite first, then convert by nncase

I have tried, but because the shape of onnx is (N, C, H, W) but tf is (N, H, W, C), so occur this error when use `ncc` to convert.

Can you contact with nncase's developer and try to fix this?

I have been tortured by this for three weeks, no matter how I post issue, there still has bugs.

```sh
1. Import graph...
2. Optimize Pass 1...
3. Optimize Pass 2...
4. Quantize...
  4.1. Add quantization checkpoints...
  4.2. Get activation ranges...
  Plan buffers...
  Run calibration...
Fatal: Unsupported image channels: 240               ] 0% 0s
```

Neutree, 2020-08-14:
create issue in nncase project? https://github.com/kendryte/nncase/issues

yub1ng, 2020-08-14:
> create issue in nncase project? https://github.com/kendryte/nncase/issues

I think onnx loading error is not the nncase error, because when I use `ncc infer` the result is right

yub1ng, 2020-08-14:
> create issue in nncase project? https://github.com/kendryte/nncase/issues

Is it really not maixpy's model-parsing error?

yub1ng, 2020-09-03:
Kmodel V4 do not support kpu.net_info()

closed 2020-09-03 by yub1ng