sipeed/MaixPy-v1 #291: saving image to PC

ChahidTALHA, 2020-08-13:
I am using Maixduino board for my project, and the latest version of Maixpy IDE to develop my code. I am looking for a specific python script that would help me save images taken by the camera (OV2640) into my PC directory.
There is a way to do it by right clicking the frame buffer then save the image to a path. I want to automate this task for my project but I don't know how.
Thanks in advance.

Neutree, 2020-08-14:
image.save()

https://cn.maixpy.sipeed.com/zh/libs/machine_vision/image.html#imagesavepath-roi-quality50

closed 2020-08-14 by Neutree

ChahidTALHA, 2020-08-14:
image.save() allows me to save the image to maixduino's flash. is there a code that would allow me to save the image directly to my  machine (Windows 10)

Neutree, 2020-08-15:
use upyloader https://github.com/BetaRavener/uPyLoader/releases