sipeed/MaixPy-v1 #293: maixpy run big size flash model !

alan840706, 2020-08-16:
when i run my code:

import math
import os, sys
import sensor,image,lcd,time,network,socket
import KPU as kpu
from machine import UART
from Maix import GPIO
from fpioa_manager import fm, board_info
lcd.init(freq=10000000)
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
#sensor.set_hmirror(1)					#flip camera; maix go use sensor.set_hmirror(0)
#sensor.run(1)
#lcd.rotation(1)

clock = time.clock()

while(1):
    clock.tick()
    img = sensor.snapshot()
    while 1:
        task = kpu.load_flash (0x200000,1,0xC000,80000000)
        #code = kpu.run_yolo2(task, img)
        code = kpu.forward(task, img)
        if code:
            #for i in code:
               a=img.draw_rectangle(i.rect())
        a = lcd.display(img)
        print(clock.fps())
    

My maix go board will start and lost connection with computer  
i dont really understand   "kpu.load_flash " ,how does it work?

Neutree, 2020-08-18:
it's not completly supported, I'l update later, and you can just use kmodel V3

Neutree, 2020-08-18:
I updated firmware in latest master branch, anyu demo: https://github.com/sipeed/MaixPy_scripts/tree/master/machine_vision/load_big_model