sipeed/MaixPy-v1 #296: What version full firmware ?

anhnguyen2345, 2020-08-18:
What version full firmware can run model has 1.81M kmodel ? ( model was trained from colab with architecture mobilenet7_5)

Neutree, 2020-08-18:
customize your firmware here https://www.maixhub.com/compile.html

anhnguyen2345, 2020-08-18:
> customize your firmware here https://www.maixhub.com/compile.html
That web doesnot work for me :(. Is there full firmware available to run model 1,81M ? From here https://github.com/sipeed/MaixPy/releases .
Thank you

Neutree, 2020-08-19:
Just leave SPIFFS as default, check Modules, checkout modules you need like below:

[image]

anhnguyen2345, 2020-08-19:
When i click button in mail to download custom firmware , i got this:
[image]
And, I train model follow instruction:
https://colab.research.google.com/github/AIWintermuteAI/aXeleRate/blob/master/resources/aXeleRate_pascal20_detector.ipynb
with architecture MobileNet7_5 it generate model 1.81M
So what Module I should choose to can run this model for K210 ? Thank you.

Neutree, 2020-08-20:
just use the minimum firmware here: https://dl.sipeed.com/MAIX/MaixPy/release/master/

or select the same as the screenshot I post

anhnguyen2345, 2020-08-20:
With minimun firmware i only can run my promgram(python) via
terminal(linux), so it can not save my program if my board reset. So Can I
save program to load kmodel 1.81M whenever (sitll run program if k210
reset) ? Guide me plz. thank you.

Neutree, 2020-08-20:
the easiest way is use the online compile on maixhub, but it seems service has some problem, we will resolve it, before you can use it, you can build firmware by yourself offline.
Fllow https://github.com/sipeed/MaixPy/blob/master/build.md to build your firmware, and select IDE support in `python3 project.py menuconfig`
[image]

Neutree, 2020-08-20:
and you can download minimum with IDE support firmware
[image]

myuanz, 2020-08-20:
Hello, I've tried the service just now,  and I got a right result, could you try again in maixhub? 

> When i click button in mail to download custom firmware , i got this:
> [image]
> And, I train model follow instruction: [...]

anhnguyen2345, 2020-08-20:
Thank you, but when I run code to load kmodel 1.81 with this firmware I got
error : ValueError: [MAIXPY]kpu: load error:2002, ERR_KMODEL_VERSION: only
support kmodel V3.

Neutree, 2020-08-20:
as the error code describe, your model version is wrong, so maybe you convert your kmodel in wrong way

anhnguyen2345, 2020-08-20:
nncase 0.1.0-rc5 will convert kmodel v3, nncase 0.2.0-beta will convert
kmodel v4, is that right ? The instruction github I use, I realize It
use nncase 0.2.0-beta to convert to k210 type. I am sorry if I say wrong.
Thank you.

Neutree, 2020-08-20:
you can try again online compile tool at https://www.maixhub.com/compile.html , select kmodel v4 support, and then generate a firmware

Neutree, 2020-08-20:
or you can directly use V3 model, and we recommend you use V3 kmodel if your model don't have to use V4, it's faster and save more memory

anhnguyen2345, 2020-08-20:
Thank you, Sadly, I tick module Kmodel V4 and submit, I get a link in my
mail, but when I click I got error again.

myuanz, 2020-08-20:
> Thank you, Sadly, I tick module Kmodel V4 and submit, I get a link in my mail, but when I click I got error again. Vào Th 5, 20 thg 8, 2020 vào lúc 16:35 Neucrack <notifications@github.com> đã viết:
> […](#)

Could you show your config or try againinin? I have cleared all cache.

koda87, 2023-06-20:
There is another link for the online build too? This link not work https://www.maixhub.com/compile.html