sipeed/MaixPy-v1 #299: Maixpy Error 13 Help !

Demet95, 2020-08-24:
Hey everbody. I followed a racoon detector tutorial  and when i tried to test it i just got and error in the code.The kmodel is fine i think but for some reason i just got:
Traceback (most recent call last):

  File "<stdin>", line 21, in <module>
OSError: run error: 13
MicroPython v0.5.0-29-g97fad3a on 2020-03-13; Sipeed_M1 with kendryte-k210
Type "help()" for more information.

I follow this script :https://github.com/AIWintermuteAI/aXeleRate/blob/master/example_scripts/k210/detector/racoon_detector.py

and just modify the While cycle due an error with these lines :
.rotation_corr(z_rotation=90.0)
    a = img.pix_to_ai()

I'll really appreciate any help.

Neutree, 2020-08-24:
show your full code

nlothian, 2020-08-24:
It's pretty likely you don't have your model loaded at `0x200000`. See this line: `task = kpu.load(0x200000)`

You need to either load the model from the (easy method) SD card or flash the model to the boards memory at the correct memory point.

Demet95, 2020-08-24:
> show your full code

[image]

Neutree, 2020-08-24:
@Demet95  try the latest firmware in master branch https://dl.sipeed.com/MAIX/MaixPy/release/master/

Demet95, 2020-08-24:
@Neutree  Hey now im just got this error when i used the latest firmware:
[image]

Neutree, 2020-08-25:
visit https://www.maixhub.com/compile.html , and compile a support v4 model firmware

[image]

Demet95, 2020-08-25:
@Neutree Hi , i just did it and no errors where found but, the code it's not working and the other strange thing is that the MaixGo got disconnecting from de IDE by itself. This is what the terminal showed me :
[image]

Neutree, 2020-08-25:
use terminal tool to connect board, and run

Demet95, 2020-08-25:
@Neutree  Hi i did it and showed me this:
[image]
I don't know what the problem is, but i suppose that it's because the k model?

Neutree, 2020-08-26:
it maybe a bug, but you should show your firmware's version

Neutree, 2020-08-26:
and it's better put your model here if you can share it

Demet95, 2020-08-26:
Hey @Neutree  these are my model and firmware as u request. I really apreciate all the help that you are giving to me

[Model.zip](https://github.com/sipeed/MaixPy/files/5127727/Model.zip)

Neutree, 2020-08-26:
I tried this model with latest commit, it can be load, but memory seems not enough,

```
#tested with frimware 5-0.22
import sensor,image
import KPU as kpu

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.set_windowing((224, 224))
sensor.set_vflip(1)
sensor.run(1)
classes = ["racoon"]
task = kpu.load(0x300000) #change to "/sd/name_of_the_model_file.kmodel" if loading from SD card
a = kpu.set_outputs(task, 0, 7,7,30)   #the actual shape needs to match the last layer shape of your model(before Reshape)
anchor = (0.57273, 0.677385, 1.87446, 2.06253, 3.33843, 5.47434, 7.88282, 3.52778, 9.77052, 9.16828)
a = kpu.init_yolo2(task, 0.3, 0.3, 5, anchor) #tweak the second parameter if you're getting too many false positives
while(True):
    img = sensor.snapshot().rotation_corr(z_rotation=90.0)
    a = img.pix_to_ai()
    code = kpu.run_yolo2(task, img)
    if code:
        for i in code:
            a=img.draw_rectangle(i.rect(),color = (0, 255, 0))
            a = img.draw_string(i.x(),i.y(), classes[i.classid()], color=(255,0,0), scale=3)
a = kpu.deinit(task)
```

[maixpy.bin.zip](https://github.com/sipeed/MaixPy/files/5128512/maixpy.bin.zip)

Demet95, 2020-08-26:
I run the script again (from the terminal) with the firmware or uploaded in the zip file. I just got what the image says, and i really don't understand why the mismatch between the shapes , i just followed this tutorial: https://www.youtube.com/watch?v=87c6dCgXeJo&lc=UgwRad8j1nV94reFwz94AaABAg.9CiZCaIO5QL9CmEylVQAZI

[image]

AIWintermuteAI, 2020-08-31:
If you are following the tutorial, the shape should match - you can see it during the training
The problem you're experiencing is referenced in this issue:
https://github.com/AIWintermuteAI/aXeleRate/issues/10

Demet95, 2020-08-31:
Hi @AIWintermuteAI  i tried what u post me and I could fix that error. However, i got this :
[image]
line 25 refer to :             a = img.draw_string(i.x(),i.y(), classes[i.classid()], color=(255,0,0), scale=3)

Can you help me? I don't know if this is cause the firmware. Other thing to consider i think is that in the screen of the board  i got this :
[image]
The red part is the initialization of the IDE firmware , looks like a bug but i dont really know. After a few seconds the code is running goes the above error.

AIWintermuteAI, 2020-09-01:
What is your model? It is not racoon detector, isn't it?
Seems like the number of classes you specified here
> classes = ["racoon"]

does not match the number of classes output by the model. 
Thus my question - what is your model? If you modified the model, then make sure you spend time to understand how to modify the code and change the relevant parts.

Demet95, 2020-09-01:
U r so right and I'm so ashamed that i did not even noticed it. Im new into this world and im learning the basics yet. But with all these mistakes i think ill improve. 
I have one more question if u allow me : Do u think is possible to do an ANPR on the Maix?

I could run both's models , thanks so much for help.

Demet95, 2020-09-01:
@AIWintermuteAI  U r so right and I'm so ashamed that i did not even noticed it. Im new into this world and im learning the basics yet. But with all these mistakes i think ill improve.
I have one more question if u allow me : Do u think is possible to do an ANPR on the Maix?

I could run both's models , thanks so much for help