sipeed/MaixPy-v1 #300: Unable to load kmodel v4

shanejohnpaul, 2020-08-25:
Firmware: maixpy_v0.5.0_123_g859098d
NNcase: v0.2.0 Beta4

I saw this [issue ](https://github.com/sipeed/MaixPy/issues/182)and it said to use the latest firmware to support kmodel v4. 

This is the error I got.
`ValueError: [MAIXPY]kpu: load error:2002, ERR_KMODEL_VERSION: only support kmodel V3`

Neutree, 2020-08-26:
https://dl.sipeed.com/MAIX/MaixPy/release/master/maixpy_v0.5.0_124_g92d9c04

[image]

zyx-dev, 2020-08-26:
whether the latest version “with lvgl” support kmodel V4?

Neutree, 2020-08-26:
@zyx-dev  upload failed, but you can customize your firmware here: https://www.maixhub.com/compile.html

shanejohnpaul, 2020-08-26:
> https://dl.sipeed.com/MAIX/MaixPy/release/master/maixpy_v0.5.0_124_g92d9c04
> 
> [image]

This actually worked!!! (the one with model v4 support) But now I can only load my model. When I try to set the frame size of the camera to QVGA, it says `E (22399959088) SYSCALL: Out of memory`.

The model is just 2.25MB and I downloaded it at 0x200000. Is there something we could do?

Neutree, 2020-08-26:
compile your firmware follow the build.md, remove the module you don't need, and try to reduce the lcd resolution
[image]

and if you can kmodel V3, just use V3, and V3 support kpu.load_flash(), which support load big model from flash when running

Neutree, 2020-08-26:
[image]

[image]

shanejohnpaul, 2020-08-26:
Thanks. I'll try them.

shanejohnpaul, 2020-08-26:
Does the online compiler work? I chose the kmodel v4 option in it and it wasn't able to load the model.

shanejohnpaul, 2020-08-26:
I compiled the firmware as mentioned in [here](https://github.com/sipeed/MaixPy/blob/master/build.md). I went with the maixpy_k210_minimum project. I set all the options as mentioned above.  But still, it's running out of memory when trying to load the camera sensor. Is there anything else I could do?

I tried to compile my tflite model to kmodel v3 but I'm getting some errors. But with v4, everything went fine.