sipeed/MaixPy-v1 #301: What functions are available while using the minimum firmware ?

staberas, 2020-08-25:
What functions are available for the image & Kpu while on the minimum firmware ? 
Im trying to find information and i cant find anything... Im currently trying to see my options since i need the face recognition model and find_similarity ,  find_blobs , QR and _thread for my project. Thanks

Neutree, 2020-08-26:
just use the normal firmware, minimum no these function, and you can customize your firmware here: 
https://www.maixhub.com/compile.html

AIWintermuteAI, 2020-08-26:
Just to expand on the answer - if you build the firmware manually and open the config file for minimum firmware you will see this:
[image]
So, find blobs, QR and thread are not available in minimum firmware.

staberas, 2020-08-26:
Thanks, but i have used the custom firmware creator  ( min omv & _thread )  but my code still crashes. Anyway to make my question more clear is , which functions min omv  has available to use?

Neutree, 2020-08-26:
basic image ops like `draw_...` `copy` etc. but no algrightm like `find_...`

staberas, 2020-08-26:
Found it:  booting the maix without an sd into REPL and then typing ` >>>help(image.Image())` 
or  `>>>help(kpu) ` etc and it will give a list of functions that are available.

closed 2020-08-26 by staberas