sipeed/MaixPy-v1 #304: Firmware v0.5.0_200 & v0.5.0_202	have jpg viewing problem

staberas, 2020-08-31:
Just flashed maixpy_v0.5.0_200_g8f6aa7b & maixpy_v0.5.0_202 (minimum both) and while they solve some problems i had the jpg images that are displayed with incorrect coloring... and im also thing there is a scaling issue

staberas, 2020-09-01:
[image]
there is also extra space on the top and bottom

junhuanchen, 2020-09-02:
Maybe you could test tft_fill_data to determine how much the size of its entire area is different.

https://github.com/sipeed/MaixPy/blob/e1ebb56db6457b1a4da072ed5cba44e22b6be06a/components/drivers/lcd/src/st7789.c#L133

Check the situation.

staberas, 2020-09-02:
did lcd.clear((255,206,0)) and its filling it with yellow color 
note the pictures are 320x228 and previous firmwares worked fine. I can see the latest firmware has applied some rescaling since now the text now is way less blocky that the previous versions

junhuanchen, 2020-09-02:
There was an overflow of the previously filled address, so this problem was not found.

In particular, there are undefined bugs in the area of rotation.

junhuanchen, 2020-09-02:
Can you see this https://github.com/sipeed/MaixPy/commits/master/components/drivers/lcd/src/lcd.c

It is best if the cause of the problem is identified before and after

junhuanchen, 2020-09-02:
Maybe you can show 320 by 240 for comparison

staberas, 2020-09-02:
Ok, this is interesting i initialised the lcd way after i did other stuff (like loading the models, init servo etc ) it seems now that i moved the init of lcd at the start the color problem has vanished ...

junhuanchen, 2020-09-03:
Previously, the starting address in py_lcd.c was not initialized, which caused a lot of strange phenomena. If there are similar situations, you can attach the code + picture and reopen the issue.
THX ~

closed 2020-09-03 by junhuanchen