sipeed/MaixPy-v1 #305: Wrong EPOCH reference time

Pali775, 2020-08-31:
Using Maixduino_esp32_fimware v1.4.1 and MicroPython v0.5.0-200-g8f6aa7b, the EPOCH reference time is wrong.

As per [utime class](https://docs.micropython.org/en/latest/library/utime.html) the "_time Epoch: Unix port uses standard for POSIX systems epoch of 1970-01-01 00:00:00 UTC. However, embedded ports use epoch of **2000-01-01 00:00:00 UTC**._"

```
...
# reference time (in seconds)
SHIFT = 3155666400

try:
    res = s.sendto(NTP_QUERY, addr)
    data, addr = s.recvfrom(1024)
except Exception as e:
    print("receive error:", e)
s.close()

t = ustruct.unpack("!12I", data)[10]
t -= SHIFT
print(utime.localtime(t))
...
```
Value **3155666400** to shift the timestamp seems to be a mistake.

Neutree, 2020-09-04:
micropython use the 2000 instead of 1970: https://github.com/micropython/micropython/blob/073b9a5eb888b26be7ce4c65e9649b6c3857ba39/ports/stm32/modutime.c#L120-L129

but maixpy use 1970 in `time.time()` and use 2000 in `utime.localtime`, it's a bug, we should change it to 2000 in `time.time()`

Neutree, 2020-09-04:
you should set SHIFT to `946684800` if you want to convert to UTC

Neutree, 2020-09-05:
change to start from 2000 instead of 1970 now, keep the same as micropython do

d1c6886e53048ae712fb07d1f0724c23d0f903b7

closed 2020-09-05 by Neutree