sipeed/MaixPy-v1 #307: ESP32 status function is not available

Pali775, 2020-09-13:
Using "MicroPython v0.5.0-200-g8f6aa7b on 2020-08-31; Sipeed_M1 with kendryte-k210" on board: Maixduino

There is no status function for NIC created as ESP32_SPI that seems a bit strange as the function is defined in esp32\modnetwork.c:

```
STATIC mp_obj_t esp_status(size_t n_args, const mp_obj_t *args) {
    wlan_if_obj_t *self = MP_OBJ_TO_PTR(args[0]);
    ...
    return mp_const_none;
}
```
Not only the function, but the Python reference is defined too:
`STATIC MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(esp_status_obj, 1, 2, esp_status);`

Also given among the properties:
```
STATIC const mp_rom_map_elem_t wlan_if_locals_dict_table[] = {
    { MP_ROM_QSTR(MP_QSTR_active), MP_ROM_PTR(&esp_active_obj) },
    { MP_ROM_QSTR(MP_QSTR_connect), MP_ROM_PTR(&esp_connect_obj) },
    { MP_ROM_QSTR(MP_QSTR_disconnect), MP_ROM_PTR(&esp_disconnect_obj) },
    { MP_ROM_QSTR(MP_QSTR_status), MP_ROM_PTR(&esp_status_obj) },
    { MP_ROM_QSTR(MP_QSTR_scan), MP_ROM_PTR(&esp_scan_obj) },
    { MP_ROM_QSTR(MP_QSTR_isconnected), MP_ROM_PTR(&esp_isconnected_obj) },
    { MP_ROM_QSTR(MP_QSTR_config), MP_ROM_PTR(&esp_config_obj) },
    { MP_ROM_QSTR(MP_QSTR_ifconfig), MP_ROM_PTR(&esp_ifconfig_obj) },
};
```
And of course, the **modnetwrok.c** is part of the **Makefile**

Why **ifconfig** or **scan** are available but **status** is not?

junhuanchen, 2020-09-13:
see this components/micropython/port/src/standard_lib/network/esp32/mod_esp32.c

Pali775, 2020-09-13:
well, it seems to be more and more messy.... not so clear the structure

anyway, I have just extended for **rssi**, what I really need:

```
STATIC mp_int_t esp32_nic_rssi( mp_obj_t self_in )
{
    return esp32_spi_get_rssi();
}

STATIC MP_DEFINE_CONST_FUN_OBJ_1(esp32_nic_rssi_obj, esp32_nic_rssi);

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

STATIC const mp_rom_map_elem_t esp32_locals_dict_table[] = {
    ...
    { MP_ROM_QSTR(MP_QSTR_rssi), MP_ROM_PTR(&esp32_nic_rssi_obj) },
    ...
};
```

I am not sure if it is the best approach, but it seems to be work

junhuanchen, 2020-09-14:
I agree with

closed 2020-09-14 by junhuanchen

voscovvo, 2024-04-08:
Why dont you do it as a standard function in next release?
the only way now to check RSSI of current connection is to do full scan, which is very slow,
Please, add get_RSSI as basic function.
Thank you.