sipeed/MaixPy-v1 #309: [M5StickV]Nothing appears on the lcd.

mongonta0716, 2020-09-15:
I have a new firmware on the M5StickV (v0.5.0-232 or later) and nothing shows up on the lcd. (The K210 and sensor are working.)

I'm reproducing this with M5StickV. (Not even with the release.)

v0.5.0-95-g05c4e7080-dirty I used to use it frequently before and it was showing correctly, I haven't checked from 96-231.
Is there any cause for this?

'''
import sensor, image, time
import lcd

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)
lcd.init()
clock = time.clock()

while(True):
    clock.tick()
    img = sensor.snapshot()
    a = lcd.display(img)
    print(clock.fps())
'''

mongonta0716, 2020-09-15:
My verification does not show the M5StickV lcd after [this commit](https://github.com/sipeed/MaixPy/commit/a3a5d1a1975fa19265809d6b81d2ea02e91ce09f).

However, even if I revert this commit only, it doesn't work.

junhuanchen, 2020-09-16:
Well~
Maybe I need a M5StickV...

junhuanchen, 2020-09-16:
Okokok, I will check it. QAQ

mongonta0716, 2020-09-16:
Please.

I don't think it's a fix that would stop the Lcd from showing, but I couldn't figure out why.

I even looked up the fact that if I checkout to [this commit](https://github.com/sipeed/MaixPy/commit/01c55e384ff360835d4bc1157cf8bd3a9575bfe0), it will show up.

The Lcd backlight is on and nothing is being drawn.

junhuanchen, 2020-09-16:
see this https://github.com/sipeed/MaixPy/commit/69980347e19482cf3713766919822a4a276aeadc

In the past, the macro definition was compiled into the firmware, now lcd.init(type=xx) is used.

closed 2020-09-16 by junhuanchen

mongonta0716, 2020-09-16:
It worked fine.

Thank you.