sipeed/MaixPy-v1 #310: [M5StickV]SD not working

mongonta0716, 2020-09-19:
The TF Card doesn't work as well as the Lcd stopped working in #309.

This is the commit that stopped working.
a3a5d1a1975fa19265809d6b81d2ea02e91ce09f

If I check out to this commit, it works.
01c55e384ff360835d4bc1157cf8bd3a9575bfe0

junhuanchen, 2020-09-21:
TF Card's comparison schematic.

Can you confirm the hardware difference?

mongonta0716, 2020-09-21:
I can use it when I put the commit back in.

It's a TF Card card with no problems. I checked the source and it looks like it doesn't support the M5StickV GPIOs after the relevant commit?

My TF Card
Lexar 32GB A1 V10

junhuanchen, 2020-09-21:
It seems that there is no problem with the definition, maybe you need to confirm whether the sd card is working, and whether it is related to lcd.init().

https://github.com/sipeed/MaixPy/blob/820add96a6f7048f328bc9778bd91188ef248005/components/drivers/sd_card/src/sdcard.c#L428-L446

https://github.com/sipeed/MaixPy/blob/820add96a6f7048f328bc9778bd91188ef248005/components/micropython/port/src/omv/py/py_lcd.c#L284-L299

mongonta0716, 2020-09-21:
I'm seeing the phenomenon on three M5StickVs I have on hand.
A few more people have confirmed the same phenomenon.

Can your M5StickV read?

junhuanchen, 2020-09-21:
The SD card slot on my hand is broken

junhuanchen, 2020-09-21:
I am also very confused. Maybe the pins and functions of the SD card slot should be confirmed separately.

mongonta0716, 2020-09-21:
My ISSUE was poorly written.

It has nothing to do with the LCD.
It means that the TF Card cannot be read after the same commit.

Does the fact that you're broken mean you can't check out the commit?

junhuanchen, 2020-09-21:
I cannot confirm that my M5StickV  SD card is a software problem

mongonta0716, 2020-09-21:
Can't you figure out...

I'll do some more research on my own.

The point I'm wondering about is that CS_PIN is not 29 with the M5StickV diagram. However, if I set it to 32, it doesn't work at all...

mongonta0716, 2020-09-21:
This might have fixed it.

I'll test it some more.

[image]

junhuanchen, 2020-09-21:
Has been resolved

closed 2020-09-21 by junhuanchen