sipeed/MaixPy-v1 #317: kpu.load or kpu.load_flash failed

rainnies, 2020-09-30:
hi, 
When i run script below, i always got those error:
[image]
When i use "kpu.load_flash", i got "**ERR_KMODEL_VERSION: only support kmodel V3/V4 now**".
Use "kpu.load" will get "**E (890188367114) SYSCALL: Out of memory**".
The model is convert by nncase V0.2.0, kmodel size is about 1.9 mb. 
firmware version I've try:
maixpy_v0.5.1_110_g450f20b_minimum_with_kmodel_v4_support.bin
maixpy_v0.5.1_110_g450f20b_minimum
maixpy_v0.5.1_110_g450f20b.bin

Here is the model via Yolo-digit-detector and convert via nncase V0.2.0:
[raccoon1.kmodel.zip](https://github.com/sipeed/MaixPy/files/5304109/raccoon1.kmodel.zip)

Neutree, 2020-10-14:
download the firmware with kmodel v4 support verstion

https://dl.sipeed.com/MAIX/MaixPy/release/master/

[image]

michaelteo, 2020-12-13:
Is kpu.load_flash() already supporting kmodel v4? I saw in another thread that kmodel v3 can use load_flash. (https://github.com/sipeed/MaixPy/issues/300#issuecomment-680755127)  

I tried to load_flash a kmodel v4  using the newest firmware - maixpy_v0.6_1_gfa51290 - but I am still getting the error:
ValueError: [MAIXPY]kpu: load_flash error:2002, ERR_KMODEL_VERSION: only support kmodel V3/V4 now

The file size is just 1.7MB
[handlandmark.zip](https://github.com/sipeed/MaixPy/files/5684324/handlandmark.zip)

But when I compiled in nncase it said it: Working memory usage: 3010560 B. This is failing for kpu.load(). Based on trial and error - It looks like the working memory is the size that matters not the file size. 

The only way I was successful with kpu.load() was to make the kmodel smaller by quantization ( down to 1.5MB working memory ) AND also using firmware which are "minimum with kmodelv4". But the result was very inaccurate and it was not useful.

michaelteo, 2020-12-25:
@Neutree - 
It seems there is an issue with kpu.load_flash() in firmware "maixpy_v0.6_9_g88bfd0177_openmv_kmodel_v4_with_ide_support.bin"

I am getting an error "ValueError: [MAIXPY]kpu: load_flash error:2006, ERR_NO_MEM: memory not enough"
But it should not be the case since the model I am loding from flash is 3.7MB only so within the range of flash memory (16MB )

michaelteo, 2020-12-25:
@Neutree - I tried with new firmware, but seems there is still an issue loading from flash for kmodel v4. 

Firmware used: "maixpy_v0.6.2_5_g7962bda89.bin"
KModel Version : Kmodel4 converted from TFlite using NNCase v0.2.0 Beta4

To create the model, I converted from TFlite to kmodel using NNCase v0.2.0 beta4. Then I ran the convert_le.py to make the kmodel be little endian as per this guide: https://en.bbs.sipeed.com/t/topic/1811

When I tried to load using kpu.load_flash(), I am not getting any errors, but when I print the task, it says model size = 0 

```
{"model_addr": 3145728, "model_size": 0, "model_path": "(null)", "net_args": "(null)"}
w*h*ch!=0
[MAIXPY]kpu: set_outputs arg value error: w,c,ch size not match output size
w*h*ch!=0
[MAIXPY]kpu: set_outputs arg value error: w,c,ch size not match output size
w*h*ch!=0
[MAIXPY]kpu: set_outputs arg value error: w,c,ch size not match output size
```

here is the code segment I used:
```
task_he = kpu.load_flash(0x300000,0,0x4000,40000000)
print(task_he)
a=kpu.set_outputs(task_he, 0, 63,1,1)
a=kpu.set_outputs(task_he, 1, 1,1,1)
a=kpu.set_outputs(task_he, 2, 1,1,1)

```

Attaching models 
[models.zip](https://github.com/sipeed/MaixPy/files/5742101/models.zip)

I tried the "mbnet75_3.flashrun.kmodel" in the above guide. It seems to be loading from flash using this firmware ( it is reporting modelsize >0). But that model might still be kmodelv3. 

Other observations - when i tried SPI of 80000000, i get this error for exact same firmware, code and model. But when it's 60000000 and below, the error is gone and I get the issue above. 

`ValueError: [MAIXPY]kpu: load_flash error:2002, ERR_KMODEL_VERSION: only support kmodel V3/V4 now
`

KuoEuran, 2021-07-10:
@michaelteo
 Hi, have you solved the issue? I am facing the issue when I use kpu.load_flash.
[image]
[image]
And it also shows flash memory is not enough
[image]
And I use the SD card to load my komel, it shows
[image]

I don't understand what happened. Can you give me some advice about it? tks.

michaelteo, 2021-07-10:
I haven't worked on this in long while and didn't get it working. 

But some ideas - maybe setting the is_dual_buf to 0 might help on memory not enough error, or setting batch size lower to 0x4000. 

for loading from SD - maybe try the original model before converted to Big Endian?

KuoEuran, 2021-07-10:
@michaelteo Thank for your replying, unfortunately, it is still memory not enough error.

rainnies, 2021-07-10:
I used v1 nncase and it works perfectly. I gave up kmodel v4 finally.

KuoEuran, 2021-07-11:
@rainnies Hi, thank you for your reply
You mean you use v0.1 nncase convert your model to kmdel, and work fine on the board?
What's your .bin file you use? 
Can you use the command like kpu.load_flash or info_list = kpu.netinfo(task) perfectly?