sipeed/MaixPy-v1 #318: kpu.run_yolo2 breaks ws2812 library

AndreiUngureanu101, 2020-10-01:
Hello,

I recently discovered a weird issue regarding the maix bit.  

I have a `ov2640` camera and some `ws2812` leds connected to it. I start by loading a model for inference using `kpu.load`. If I then run the `ws2812.set_led` method for lighting the leds everything is working fine, but if I run the `kpu.run_yolo2` after doing the `kpu.load` and then try to light the leds using `ws2812.set_led`, nothing is happening with the leds. It's as if the `kpu.run_yolo2` makes the pin communication inactive.

Is there any reason for this behaviour? Thanks!

junhuanchen, 2020-10-14:
Previous firmware has had DMA conflicts, now you can test on the latest firmware.

https://github.com/sipeed/MaixPy/blob/22ab99ea284a63394f6e1123e90af6cb7d4c2ea6/components/micropython/port/src/modules/ws2812/src/py_ws2812.c#L46

If there are still problems, you can rule them out by setting the parameters here.

It comes with your test code.

AndreiUngureanu101, 2020-10-22:
It seems to be working now, thanks!

closed 2020-11-13 by junhuanchen