sipeed/MaixPy-v1 #319: Wrong pin numbers for Amigo + How to set flash led on?

AngainorDev, 2020-10-04:
https://github.com/sipeed/MaixPy/blob/f9bb0bb5e807846e6b6e397b9ced2963c5472fab/components/micropython/port/builtin_py/board.py#L16

On my Amigo board, only LED_R is the right pin (14)  
LED_G is on pin 15  (and not 13)
LED_B is on pin 17  (and not 12)

How is this supposed to work if the maixpy is the same for every boards?

Also, what is the pin number of the white led (rear flash)?
The schematic does not help, I tried several unlabeled pins without success.

AngainorDev, 2020-10-04:
White Led is on pin 32 on my Amigo.

junhuanchen, 2020-10-04:
see this https://github.com/sipeed/MaixUI/blob/5646aa3899d126e579e99d38bb8020857cd3abe3/driver/led.py#L37

AngainorDev, 2020-10-04:
Thanks, found it looking at MaixUI precisely (MaixUI demo not working on Amigo either btw, tried several firmwares with no luck)

But if I got it right, nothing is ready yet for Amigo as far as code and doc is concerned?
MaixPy constants and board definitions are not right, lvgl touch screen is not working, demo scripts are not compatible for all sorts of reasons aso...
It's really hard to get started, one has to sort and re-invent all missing and misleading doc.
(even the chinese doc is not up to date)

nlothian, 2020-10-05:
@AngainorDev see https://github.com/sipeed/MaixPy/issues/298 where they say they will ship board configuration files for each board "later" (unsure when that is)

rolland-fx, 2020-10-06:
But if I got it right, nothing is ready yet for Amigo as far as code and doc is concerned?

I'm afraid of the same things, most library don't work out of the box for now.
The current up to date firmware does not allow touchscreen, or even to capture pictures to the resolution of the screen from sensors.
Hopefully, it will be supported quickly, since this hardware + a working Maixpy will be a very good device to work on.

AngainorDev, 2020-10-06:
Agreed.
It also looks like SiPeed relies a lot on Open source and us -  os devs - to get it rolling.

I started https://github.com/AngainorDev/Maix-Amigo-Help
to collect my notes and snippets, feel free to submit yours.

I don't want that nice platform to die for lack of doc/code!

junhuanchen, 2020-10-07:
> @AngainorDev看到[＃298](https://github.com/sipeed/MaixPy/issues/298)，他们说他们将“以后”为每块板运送板配置文件（不确定何时）

For example, this code https://github.com/sipeed/MaixUI/blob/master/config_cube%26amigo.py

junhuanchen, 2020-10-07:
at present, amigo has not started to organize documents and sample codes.

but all functions and tests are done on MaixUI.

https://github.com/sipeed/MaixUI/tree/master/driver

This is currently not documented, only the code tested.

junhuanchen, 2020-10-13:
now ，can use this https://github.com/sipeed/MaixPy_scripts/tree/master/board

closed 2020-11-13 by junhuanchen