sipeed/MaixPy-v1 #320: Invert UART communication

bhorvathwb, 2020-10-07:
Hi!

I am using maixpy on a M5StickV K210 chip camera. For a project I would need to invert the UART communication. I have found solutions for this on the original MicroPython documentation (the _invert_ keyword parameter on the _UART.init_ function) but it throws an Extra keyword argument TypeError for me.

Here is the link to the documentation:
https://docs.micropython.org/en/latest/library/machine.UART.html

Is this possible in the MaixPy environment?

robert-hh, 2020-10-07:
Inverted UART mode is supported in the mainline MicroPython not for all ports. AFAIK only for the ESP32 port. This feature hase to be supported by the underlying platform. If this is true for the SiPeed platform, then it could be implemented. Otherwise you have to use external inverters.

Neutree, 2020-10-14:
k210 can do this by fpioa, but we not implement this on maixpy