sipeed/MaixPy-v1 #321: Colors inverting after any FW download

Kabron287, 2020-10-07:
Board MaixCube with v0.5.0-95-g05c4e70 out of the box.

if I download the same FW maixpy_v0.5.0_95_g05c4e70.bin or any newvest e.g. maixpy_v0.5.1_103_gf9bb0bb.bin,
I got inverted colors.
And this is not the first case. I got similar results with other boards.
Fortunatelly, now I did it to RAM, instead of flash.

What could be done?

junhuanchen, 2020-10-13:
see this https://github.com/sipeed/MaixPy_scripts/tree/master/board

Kabron287, 2020-10-14:
> see this https://github.com/sipeed/MaixPy_scripts/tree/master/board

Thanks a lot. It's important update.

closed 2020-11-13 by junhuanchen

Kabron287, 2021-01-11:
After upgrade to v0.6.2_12 colors returned to **inverted** state. **config_maix_cube.py** does not helps.

Kabron287, 2021-01-11:
It was due wrong example code here:
https://maixpy.sipeed.com/zh/develop_kit_board/maix_cube.html#