sipeed/MaixPy-v1 #327: Out of memory error (HEAP) in frame differencing example

Milan991, 2020-10-12:
Can you send me an example of `image.Image` usage because when I'm using instead of `extra_fb = image.Image(size=(320, 240))` I'm getting out of heap memory error.

Even when I lower the resolution to 176, 120 error remains.

My goal is to use this openMV example: `in_memory_basic_frame_differencing.py`

Board is MaixCube.

junhuanchen, 2020-10-13:
provide this code `in_memory_basic_frame_differencing.py`

Milan991, 2020-10-13:
@junhuanchen [https://github.com/openmv/openmv/blob/master/scripts/examples/20-Frame-Differencing/in_memory_basic_frame_differencing.py](https://github.com/openmv/openmv/blob/master/scripts/examples/20-Frame-Differencing/in_memory_basic_frame_differencing.py)

I'm using `image.Image(size=(320, 240))` instead of `sensor.alloc_extra_fb(sensor.width(), sensor.height(), sensor.RGB565)`