sipeed/MaixPy-v1 #330: Building firmware with custom C/C++ code

AndreiUngureanu101, 2020-10-19:
Hello,

Has anyone built the firmware with added C/C++ functionality? In theory (as mentioned here: https://github.com/Neutree/c_cpp_project_framework) it should work and even though my custom library is added in the build step I can't seem to import it, it's unavailable.

Thanks!

phquanta, 2020-10-22:
We had to rewrite pix_to_ai functionality with port to MicroPython due to memory leakage with custom resizing.

AndreiUngureanu101, 2020-10-22:
> We had to rewrite pix_to_ai functionality with port to MicroPython due to memory leakage with custom resizing.

So this is a reason for it not working?

phquanta, 2020-10-22:
_> > We had to rewrite pix_to_ai functionality with port to MicroPython due to memory leakage with custom resizing.
> 
> So this is a reason for it not working?

No, the reason for this was to crop the image to a custom sized ML model. Funny thing is that image.Image does not have structure pixai or whatever it is called when transforming image to 8 bit  AI representation. sensor.snahpshot on the other hand has such structure and if you want to resize sensor.snapshot it returns image without pixai structure and you have to manually call pix_to_ai. If you have resizing and pix_ai calls they both use malloc every time, and you have to garbage collect those quite fast. We had exhausted memory on the second pass of infinite while loop.  Basically, we had to create buffered pix_to_ai call which does not call malloc every time but rather takes custom created image and stores results there without destroying it afterwards.

AndreiUngureanu101, 2020-10-22:
Good thing you mentioned this, is there a build with this pix_to_ai functionality working properly? I tried a few and I'm always getting the "MemoryError: Out of normal MicroPython Heap Memory! " error after a couple of images loaded from sd

phquanta, 2020-10-22:
> Good thing you mentioned this, is there a build with this pix_to_ai functionality working properly? I tried a few and I'm always getting the "MemoryError: Out of normal MicroPython Heap Memory! " error after a couple of images loaded from sd

Did you try to increase heapsize ? Are you getting those errors on images from SD or from camera through sensor.snapshot() ?

AndreiUngureanu101, 2020-10-23:
I increased the heapsize and it resolved the issue, thanks! I was reading images saved in the SD card.

wetfgewt, 2024-05-30:
> 由于自定义调整大小导致内存泄漏，我们不得不使用 Port 重写 MicroPython 的pix_to_ai功能。

请问一个问题，我在ubantu编译完maixpy_k210这project之后，把生成的maixpy.bin烧录在k210 M1n里，但是一直无法连接MaixPy IDE，之后就提示固件超时错误。
[image]