sipeed/MaixPy-v1 #336: Camera rotation promble and ¿OpenCV?

Demet95, 2020-10-22:
Hi , Im working on a model in object detection (vehicular plates). I have two questions, first when I upload the base code to the maix it works perfectly but I have two Issues : First, the camera is always in rotation mode and I dont know how to make it stay fixed (The code I use is the below). The other part is that when I disconnect the Maix from the computer, the image freezes after a while.
[image]

The second question is about once I have the detection I want to recognize the letters. It is possible to install OpenCV/tesseract to do this? or What can I do? 
I'll really appreciate any help.

Neutree, 2020-11-23:
for camera you can call`sensor.set_vflip(True)` and `sensor.set_hmirror(True)` to adjuge direction, but can not change rotation, you can only put the hardware to the right direction or change anothor direction's sensor, and you can also rotate image when display on lcd by `lcd.rotation()`

Neutree, 2020-11-23:
No openCV support, you can train model like MNIST to recognize letter like this https://blog.sipeed.com/p/668.html