sipeed/MaixPy-v1 #342: how kpu.face_compare works exactly?

sddwt, 2020-11-06:
I used cosine distance between vector record_ftrs and features.
But it get changed rapidly and not stable as function kpu.face_compare works, and it also can not calculate well,
So I want to know how this function works? Could you just tell me how to achieve it by myself not use micropython?
I can't find it on resource code.
Thank you.

michaelteo, 2020-12-15:
I had the same question as well what kpu.face_encode and kpu.face_compare - there doesn't seem to be documentation around it. 

I tried to look around in the code in the repo and this is the closest reference I could see.
https://github.com/sipeed/MaixPy/blob/master/components/micropython/port/src/Maix/Maix_kpu.c

From there, one can see the core is from these two functions 
- sipeed_kpu_face_encode(float* feature, int8_t* compress_feature, uint32_t len);
- sipeed_kpu_face_compare(int8_t* feature0, int8_t* feature1, uint32_t len);

Based on my very limited knowledge of C - it looks like it is picking up from the following header.
https://github.com/sipeed/MaixPy/blob/master/components/kendryte_sdk/include/sipeed_kpu.h

Unfortunately it looks like the source code for those implementations are not in repo but are just linked to compiled object lib_sipeed_kpu.a or lib_sipeed_kpu_v4.a

But based on the structure of the functions, it looks like it is converting the floats to int8_t - maybe doing some kind of quantization before doing similarity?  I assume similarity search is also optimized to work on that converted int8_t

sddwt, 2020-12-19:
> I had the same question as well what kpu.face_encode and kpu.face_compare - there doesn't seem to be documentation around it.
>
> I tried to look around in the code in the repo and this is the closest [...]

Hi,
I alreay used feature vector as cosin similarity calculation,
the result will be a little different with compare, but it can be used.
Much better than have no func. :(

Neutree, 2020-12-20:
@sddwt right, indeed it's just cosin similarity

[image]

and then normalize result to `[0, 1]` by `*2+0.5`, finally `*100` we got probability ∈[0, 100]

closed 2020-12-20 by Neutree

michaelteo, 2020-12-21:
@Neutree - 

1) Just to confirm the normalization,  shouldn't it be `similarity / 2 + 0.5 ` since typical cos similarity is [-1,1] for non positive inputs. Otherwise the formula `similarity *2 +0.5` does will not normalize the negative ranges.

2) Also any idea to share on what `face_encode()` does? why is it needed to use for `face_compare()`? I think it is some optimization to pack/or compress the float features ?

Neutree, 2020-12-21:
right， it's a typo, sorry for that!

`face_encode` just convert the float value to range [-128, 127] according to the float value range, e.g. float range: [-0.5, 0.5] to int8 by `x/1*256`, and the float range from model output

michaelteo, 2020-12-21:
@Neutree - Thank you. Now that makes sense.  I'll try to contribute this as update to maixpy_docs later.