sipeed/MaixPy-v1 #352: Potential error in uhashlib_sha256_make_new

StefanD986, 2020-11-24:
Here
https://github.com/sipeed/MaixPy/blob/b5f626d884bdb870c33d7bff41c454c86be5d501/components/micropython/port/src/moduhashlib_maixpy.c#L38

`sha256_init(sha256_context_t *context, size_t input_len)` is called with `input_len=0xffffffff`.

with some further calculation in the [sha256_init from the K210 SDK](https://github.com/kendryte/kendryte-standalone-sdk/blob/06a2ea71f250e91d66fa156ff82ae1f5b9fc6e56/lib/drivers/sha256.c#L49) this is stored in `sha256->sha_num_reg.sha_data_cnt`.

But the `sha256->sha_num_reg`structure limits `sha_data_cnt` to a bitfield of 16 bits [here](https://github.com/kendryte/kendryte-standalone-sdk/blob/06a2ea71f250e91d66fa156ff82ae1f5b9fc6e56/lib/drivers/include/sha256.h#L34).

junhuanchen, 2020-11-26:
mark

Neutree, 2020-11-26:
fixed here https://github.com/sipeed/MaixPy/commit/a4c31becea310c90fe8e11ec1832b29dda3eb896 

but it remain can't support too large input than 3.9MiB according to the SDK's API decribe, maybe there is a the way, but we have no sha256's register datasheet, if you somebody need calculate more than 3.9MiB, can create issue in kendryte sdk for help, or dig into the source code to guess how to handle it~

closed 2020-11-26 by Neutree