sipeed/MaixPy-v1 #354: Stereo Vision example?

Rafipaulino, 2020-11-24:
I am trying to make a simple program to determine the distance at which objects are located using the dual camera for Maix Dock. 

Since OpenCV is not supported by the board, it's getting a little complicated for me to do the application. Is there an example of Stereo Vision from which to start? 

Thanks in advance

junhuanchen, 2020-11-25:
maybe you can see this https://github.com/sipeed/MaixPy_scripts/tree/master/machine_vision

Rafipaulino, 2020-11-25:
Thank you very much junhuanchen. 

I really had already checked the examples. In fact, there is a basic example to use the dual camera, but I haven't seen anything yet, for example, to calibrate the cameras and to calculate the depth of the objects.
Probably there is a way to do it using OpenMV library.

junhuanchen, 2020-11-26:
we don't provide such sample code XD， is openmv not opencv

mentarus, 2021-01-14:
Interesting I can't even see the example code that used the stereo board, I explicitly remember seeing MaixPy code using the power switch to switch between the two cameras. Is the stereo module being phased out?

mentarus, 2021-01-14:
Actually, the binocular demo is here https://github.com/sipeed/MaixPy_scripts/blob/master/hardware/demo_binocular.py it's not doing any machine vision though, as far as I can see there's no example other than just basic feed through display.

voidstring, 2022-03-31:
Object distmance example ? using binocular camera