sipeed/MaixPy-v1 #361: Error "only support kmodel V3/V4 now"

neocsr, 2020-12-14:
I'm trying to run the `demo_find_face.py` in Maix Cube with this firmware `maixpy_v0.5.1_124_ga3f708c_minimum_with_kmodel_v4_support.bin`

[demo_find_face.py](https://github.com/sipeed/MaixPy_scripts/blob/f344ef2e0038b805e61249852018a4adcae50823/machine_vision/face_find/demo_find_face.py)

This is the error:

```
> ampy run demo_find_face.py
init i2c2
[MAIXPY]: find ov7740
[MAIXPY]: find ov sensor
Traceback (most recent call last):
  File "<stdin>", line 71, in <module>
  File "<stdin>", line 66, in main
  File "<stdin>", line 63, in main
  File "<stdin>", line 63, in main
  File "<stdin>", line 50, in main
ValueError: [MAIXPY]kpu: load error:2002, ERR_KMODEL_VERSION: only support kmodel V3/V4 now
kpu_net __del__
```

I tried to use the more recent firmware `maixpy_v0.6_1_gfa51290_minimum_with_kmodel_v4_support.bin` but the Cube doesn't start after flashing that version.

This is the repl from the board:

```
[MaixPy]: soft reboot
gc heap=0x80130e20-0x801b0e20(524288)
[MaixPy] init end

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|

Official Site : https://www.sipeed.com
Wiki          : https://maixpy.sipeed.com

MicroPython v0.5.1-124-ga3f708c on 2020-09-28; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
```

Neutree, 2020-12-14:
remember to flash face find kmodel to 0x300000 (https://dl.sipeed.com/MAIX/MaixPy/model) of flash by kflash_gui

neocsr, 2020-12-14:
Thanks, that works! I wrongly assumed that the model was flashed as part of the default firmware 👍

closed 2020-12-14 by Neutree

zhaoshenhao, 2021-01-14:
I got same issue. What I have done below:

1. Loaded latest bin:
MicroPython v0.6.2-15-g0118a9a77 on 2021-01-12; Sipeed_M1 with kendryte-k210

2. Modified the code:
if __name__ == "__main__":
    try:
        main(model_addr="/sd/face_find.kmodel", lcd_rotation=0, sensor_hmirror=False, sensor_vflip=False)
        #main( model_addr=0x300000, lcd_rotation=0, sensor_hmirror=False, sensor_vflip=False)

3. Downloaded the model from https://dl.sipeed.com/MAIX/MaixPy/model

4. Upload to sd card and renamed to find_face.kmodel
>>> os.listdir('/sd')
['main.py', 'face_model_be_at_0x600000.kfpkg', 'mnist.kmodel', 'mobilenet_0x300000.kfpkg', 'face_find.kmodel']

Please advise.

Neutree, 2021-01-15:
@zhaoshenhao maybe you used a wrong model, please check your model, where's the `face_find.kmodel` come from, it should be     un compressed from `face_model_at_0x300000.kfpkg`

zhaoshenhao, 2021-01-15:
I just rename it from face_model_at_0x300000.kfpkg. How to decompress?

Neutree, 2021-01-15:
the same as zip file, just name's difference

zhaoshenhao, 2021-01-15:
@Neutree  Thank you so much. It works!