sipeed/MaixPy-v1 #362: I2C module bug

WasuMrTomass0, 2020-12-14:
Hello,
When using 'readfrom_mem' and 'writeto_mem' slave address argument that i have passed was not used. Instead it was always calling 0x77 slave.

I had to maunally change slave address to 0x77 to use those functions.

'readfrom' and 'writeto' use passed slave adresses correctly.

I have been using latest MaixpyIDE release with Sipeed M1w dock suit ( M1w dock + 2.4 inch LCD + OV2640 ) K210 Dev. Board 1st RV64 AI board for Edge Computing (https://www.seeedstudio.com/Sipeed-M1w-dock-suit-M1w-dock-2-4-inch-LCD-OV2640-K210-Dev-Board-1st-RV64-AI-board-for-Edge-Computing-p-3207.html). I was trying to communicate with MLX90640 module.

Cheers

junhuanchen, 2020-12-15:
show code & result.

use software i2c , it will have many address.

https://github.com/sipeed/MaixPy_scripts/blob/31f83ad74a74fcdc3ce90000a867feaf329320f4/hardware/demo_i2c.py#L4

WasuMrTomass0, 2020-12-15:
I have dropped this project 2 weeks ago and due to other tasks forgot to report the issue at that time. I have created two 'wrapper' functions to communicate with MLX module - the only slave on the bus.

Code below is almost the same I have been using - I changed it many times in order to find reason why it wasn't working. When I found the bug with slave address I had to drop the project anyway.

```python
from machine import I2C
i2c = I2C(I2C.I2C0, freq = 100000, scl = 28, sda = 29, addr_size = 7)
THERMAL_ADDR = 51  # 0x33

def i2c_write_data(mem_addr: int, data: int):
    i2c.writeto_mem (THERMAL_ADDR, mem_addr, data, mem_size = 8)
    pass

def i2c_read_data(mem_addr: int, nbytes: int = 2):
    return i2c.readfrom_mem (THERMAL_ADDR, mem_addr, nbytes, mem_size=8)
```

These are the functions I have created to wrap I2C comunication . When I was trying to use it I was getting an error due to no ACK bit. If I am not mistaken, it was EOF 5 or something like that.
No ACK error occured even if I was using ```writeto_mem``` or ```readfrom_mem``` functions directly in code.

I have created simple (yet not super accurate but it was working errorfree on 100kHz frequency) I2C bus listener with ESP Nodemcu to see what was happening on the bus. That was when I have noticed another slave address being transmitted. Then I tried reading to and writing from different slaves addresses to check what was going on.

Unfortunetely I don't have the module with me to reproduce the bug. However I am sure that changing slave addres with ```writeto``` (which was working fine) was a workaround for my case.

I hope it can help. Sorry I didn't report this earlier.

junhuanchen, 2020-12-16:
change I2C0 TO USE Software I2C3

hardware i2c maybe no return DATA

closed 2020-12-17 by WasuMrTomass0