sipeed/MaixPy-v1 #364: maixpy_k210 project won't build

wolfgangwayland7, 2020-12-18:
The  maixpy_k210_min builds but most in the "projects" core dump like this.. 

wolf@wolf-800-050se:~/MaxiBit/MaixPy/projects/maixpy_k210$ python3 project.py build
-- SDK_PATH:/home/wolf/MaxiBit/MaixPy
maixpy
build now
-- SDK_PATH:/home/wolf/MaxiBit/MaixPy
-- PROJECT_PATH:/home/wolf/MaxiBit/MaixPy/projects/maixpy_k210
-- Find component: /home/wolf/MaxiBit/MaixPy/components/boards
-- Find component Kconfig of boards
-- Find component: /home/wolf/MaxiBit/MaixPy/components/drivers
-- Find component Kconfig of drivers
-- Find component: /home/wolf/MaxiBit/MaixPy/components/kendryte_sdk
-- Find component Kconfig of kendryte_sdk
-- Find component: /home/wolf/MaxiBit/MaixPy/components/micropython
-- Find component Kconfig of micropython
-- Find component: /home/wolf/MaxiBit/MaixPy/components/spiffs
-- Find component Kconfig of spiffs
-- Find component: /home/wolf/MaxiBit/MaixPy/components/utils
-- Find component Kconfig of utils
-- find component: /home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/main
-- Project defaults config file:/home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/config_defaults.mk
-- python command: python3, version: Python 3.8.5

-- Load default: /home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/config_defaults.mk
-- Write makefile config at: /home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/build/config/global_config.mk
-- Write  cmake  config  at: /home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/build/config/global_config.cmake
-- write  c header file  at: /home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/build/config/global_config.h
warning: MIC_ARRAY_ENABLE (defined at /home/wolf/MaxiBit/MaixPy/components/kendryte_sdk/Kconfig:34) has leading or trailing whitespace in its prompt
-- The ASM compiler identification is GNU
-- Found assembler: /opt/kendryte-toolchain/bin/riscv64-unknown-elf-gcc
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- Check for working C compiler: /opt/kendryte-toolchain/bin/riscv64-unknown-elf-gcc
-- Check for working C compiler: /opt/kendryte-toolchain/bin/riscv64-unknown-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/kendryte-toolchain/bin/riscv64-unknown-elf-g++
-- Check for working CXX compiler: /opt/kendryte-toolchain/bin/riscv64-unknown-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Config components priority success
-- [register component: utils ], path:/home/wolf/MaxiBit/MaixPy/components/utils
-- Add V4 kpu LIB
add nncase static lib
-- [register component: kendryte_sdk ], path:/home/wolf/MaxiBit/MaixPy/components/kendryte_sdk
-- [register component: drivers ], path:/home/wolf/MaxiBit/MaixPy/components/drivers
-- [register component: spiffs ], path:/home/wolf/MaxiBit/MaixPy/components/spiffs
-- [register component: boards ], path:/home/wolf/MaxiBit/MaixPy/components/boards
-- builtin py dir: /home/wolf/MaxiBit/MaixPy/components/micropython/port/builtin_py
-- [register component: micropython ], path:/home/wolf/MaxiBit/MaixPy/components/micropython
-- [register component: main ], path:/home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/main
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/build
Scanning dependencies of target gen_mpy_cross
Scanning dependencies of target gen_version_header
Scanning dependencies of target update_build_info
Scanning dependencies of target gen_exe_src
[  0%] Generate mpy-cross ...
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
[  0%] Generating global_build_info.h ...
[  0%] Generating exe_src.c
[  0%] Built target gen_exe_src
-- Update build time and version info to header  config  at: ('/home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/build/config/global_build_info_time.h', '/home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/build/config/global_build_info_version.h')
Segmentation fault (core dumped)
make[2]: *** [micropython/CMakeFiles/gen_mpy_cross.dir/build.make:58: micropython/CMakeFiles/gen_mpy_cross] Error 139
make[1]: *** [CMakeFiles/Makefile2:520: micropython/CMakeFiles/gen_mpy_cross.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  0%] Built target update_build_info
GEN /home/wolf/MaxiBit/MaixPy/projects/maixpy_k210/build/micropython/genhdr/mpversion.h
[  0%] Built target gen_version_header
make: *** [Makefile:84: all] Error 2

Neutree, 2020-12-18:
Try run `python project.py distclean` and rebuild again

and what OS you use and OS version is?

wetfgewt, 2024-05-30:
> 尝试运行并重新生成`python project.py distclean`
> 
> 您使用的操作系统和操作系统版本是什么？

你知道怎么解决在ubantu编译完MaixPy-V1里的maixpy_k210之后，在把maxiy.bin下载到k210里连接不上MaixPy IDE的问题吗