sipeed/MaixPy-v1 #365: sensor error on Maixpy Bit with OV2640 #define OV2640_ID       (0x2642)

gprodan, 2020-12-19:
sensor error on Maixpy Bit with OV2640
I found #define OV2640_ID       (0x2642) in  MaixPy/components/micropython/port/src/omv/include/sensor.h, also defined in sensor class as 9794, but in OV2640 datasheet we have PIDH 26 and PIDL 41 (eg. 0x2641). Could this inconsistency be the cause of the error?

Neutree, 2020-12-20:
If ov2640 buy from Sipeed, it must be 0x2642, but from other place you can debug it, call `mp_printf(&mp_plat_print,  "--ID:%x", sensor->chip_id);` in function `sensro_ov_detect`(`MaixPy/components/micropython/port/src/omv/sensor.c`) to print actual ID, and build firmware to run `sensor.reset()` to show ID

gprodan, 2020-12-20:
I buy from Digiley 1597-1713-ND (Manufacturer Part Number 110991190). I build the code to print actual ID, but stil no answer from sensor. 
-------
>>> import sensor
>>> sensor.reset()
Sensor Reset-->freq:24000000
sensor_init_dvp--freq:24000000
sensor_config.i2c_num = 2
sensor_config.pin_clk = 41
sensor_config.pin_sda = 40
init i2c2
[MAIXPY]: no sensor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Reset Failed

gprodan, 2020-12-22:
My OV2640 sensor is gone (at least not answer). I change the sensor and everything works. Thanks

closed 2020-12-22 by gprodan