sipeed/MaixPy-v1 #366: video performance

hasaranga, 2020-12-21:
Hi,
How to improve the video playback performance? Is it possible to play at least 25fps?
Thanks!

Neutree, 2020-12-22:
It's totally drived by CPU, including compress JPEG and encode to MJPEG avi, so maybe some where can be improved like compress algorithm or use dual core to proccess

hasaranga, 2020-12-22:
> 
> 
> It's totally drived by CPU, including compress JPEG and encode to MJPEG avi, so maybe some where can be improved like compress algorithm or use dual core to proccess

No, I mean playback performance of the video file on sd card. I want to play 25fps video file without slowing down.

Neutree, 2020-12-22:
Right, I mean now can not play with 25fps because of the CPU performance and decode algorithm speed and  SD card read speed limitation, one of them can be a bottleneck，optimize them to achive 25fps playback or just use a higher compression rate for avi videos to reduce decode time

hasaranga, 2020-12-22:
How fast can we read from SD card? (kilobytes per second)