sipeed/MaixPy-v1 #367: Cannot build the Firmware when Speech Recognizer is Enabled

myuunosuke, 2020-12-24:
I followed instruction on how to build the firmware on Ubuntu.
I was be able to build the firmware with default configuration.
However, when I enable the "Speech Recognizer" module, it fails with following error.

Does anyone know how to solve this problem?

Thank you,

-- SDK_PATH:/home/myname/MaixPy
hello_world
build now
-- SDK_PATH:/home/myname/MaixPy
-- PROJECT_PATH:/home/myname/MaixPy/projects/hello_world
-- Find component: /home/myname/MaixPy/components/boards
-- Find component Kconfig of boards
-- Find component: /home/myname/MaixPy/components/drivers
-- Find component Kconfig of drivers
-- Find component: /home/myname/MaixPy/components/kendryte_sdk
-- Find component Kconfig of kendryte_sdk
-- Find component: /home/myname/MaixPy/components/micropython
-- Find component Kconfig of micropython
-- Find component: /home/myname/MaixPy/components/spiffs
-- Find component Kconfig of spiffs
-- Find component: /home/myname/MaixPy/components/utils
-- Find component Kconfig of utils
-- find component: /home/myname/MaixPy/projects/hello_world/main
-- Project defaults config file:/home/myname/MaixPy/projects/hello_world/config_defaults.mk
-- python command: python3, version: Python 3.6.9

-- Write makefile config at: /home/myname/MaixPy/projects/hello_world/build/config/global_config.mk
-- Write  cmake  config  at: /home/myname/MaixPy/projects/hello_world/build/config/global_config.cmake
-- write  c header file  at: /home/myname/MaixPy/projects/hello_world/build/config/global_config.h
warning: MIC_ARRAY_ENABLE (defined at /home/myname/MaixPy/components/kendryte_sdk/Kconfig:34) has leading or trailing whitespace in its prompt
-- No components priority config file
-- [register component: boards ], path:/home/myname/MaixPy/components/boards
-- [register component: drivers ], path:/home/myname/MaixPy/components/drivers
-- Add V3 kpu LIB
-- [register component: kendryte_sdk ], path:/home/myname/MaixPy/components/kendryte_sdk
-- builtin py dir: /home/myname/MaixPy/components/micropython/port/builtin_py
CMake Error at /home/myname/MaixPy/components/micropython/CMakeLists.txt:377 (get_target_property):
  get_target_property() called with non-existent target "utils".

CMake Error at /home/myname/MaixPy/components/micropython/CMakeLists.txt:377 (get_target_property):
  get_target_property() called with non-existent target "spiffs".

-- [register component: micropython ], path:/home/myname/MaixPy/components/micropython
-- [register component: spiffs ], path:/home/myname/MaixPy/components/spiffs
-- [register component: utils ], path:/home/myname/MaixPy/components/utils
-- [register component: main ], path:/home/myname/MaixPy/projects/hello_world/main
-- Configuring incomplete, errors occurred!
See also "/home/myname/MaixPy/projects/hello_world/build/CMakeFiles/CMakeOutput.log".
Makefile:279: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

Neutree, 2020-12-25:
git submodule update --recursive --init

myuunosuke, 2020-12-25:
I did execute the command and tried again, and still the same result.

There is no output after executing the command though... is it suppose to show some output messages?

username@ubuntu:~$ cd MaixPy
username@ubuntu:~/MaixPy$ git submodule update --recursive --init
username@ubuntu:~/MaixPy$

QinYUN575, 2020-12-25:
Due to historical reasons, the **helloword** project does not support many MaixPy functional components, so it is recommended to compile project **maixpy_k210**

myuunosuke, 2020-12-25:
Yes, it did solve the problem!!!

I was gonna setup another Ubuntu system to try it with clean install..., you have saved my time!

Thank you!

closed 2020-12-25 by myuunosuke