sipeed/MaixPy-v1 #368: "ImportError: cannot import name SpeechRecognizer" with custom built firmware with Speech Recognizer enabled

myuunosuke, 2020-12-25:
I have built the custom firmware with speech recognizer enabled (everything else is left as default),
and was be able to load the firmware with kflash-gui.

With the "help('modules')" command, I can see the speech_recognizer in the modules list. 

However, when I try to run the sample,
https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/speech_recognizer/speech_recognizer.py

or simply type in the import command on the monitor terminal, 
I get import error...

"ImportError: cannot import name SpeechRecognizer"

Does anyone know how to solve this problem?

Thank you,

>>> help('modules')
KPU               gc                pye_mp            uhashlib
Maix              hashlib           random            uheapq
__main__          heapq             re                uio
_boot             image             sensor            ujson
_thread           json              socket            ulab
_webrepl          lcd               speech_recognizer uos
array             machine           struct            urandom
audio             math              sys               ure
binascii          math              time              usocket
board             micropython       touchscreen       ustruct
builtins          modules           ubinascii         utime
cmath             modules           ucollections      utimeq
collections       nes               ucryptolib        uzlib
errno             network           uctypes           video
fpioa_manager     os                uerrno            zlib
Plus any modules on the filesystem
>>> from modules import SpeechRecognizer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name SpeechRecognizer
>>> import _thread
>>>

junhuanchen, 2020-12-25:
https://github.com/sipeed/MaixPy_scripts/blob/41ac9207cbf8a19be8febc1df270f4c00a2d8945/multimedia/speech_recognizer/speech_recognizer.py#L1

see https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/speech_recognizer/isolated_word.py

myuunosuke, 2020-12-25:
Ooops！
I missed that comment...

Just to be clear for the people might encounter the same problem.
"speech_recognizer.py" is no longer supported with newer firmware.
https://github.com/sipeed/MaixPy_scripts/blob/41ac9207cbf8a19be8febc1df270f4c00a2d8945/multimedia/speech_recognizer/speech_recognizer.py#L1

Please take a look at "isolated_word.py" instead.
https://github.com/sipeed/MaixPy_scripts/blob/41ac9207cbf8a19be8febc1df270f4c00a2d8945/multimedia/speech_recognizer/isolated_word.py

closed 2020-12-25 by myuunosuke

deby13, 2021-02-03:
I might be doing something very wrong, because this doesn't work either. It says isolated_word doesnt have atribute Speek

deby13, 2021-02-03:
I figured that out. It was a typo, just changed speek fot speak. that worked. thanks

NameStream, 2023-09-17:
su09-17-2023

ImportError: cannot import name MIC_ARRAY 

MAix Amigo

(1.) Hi, there.

(2.) Using MAixPy IDE, receiving error code for MAix Amigo:  

ImportError: cannot import name MIC_ARRAY 

(3.) MAix AMIGO 

(3a.) SP-MOD MicArray Module

(3b.) Connector:  10P 0.5mm FPC

(3c.) Sipeed MAix R6 + 1 Microphone Array Board

(3d.) Located inside the back of the developer board #2970, moved the switch to M1N instead of STM.  
* * * * * * * * {  By the way, STM yields same results. } * * * * * * * *

(3e.) MAixPy IDE:  Tools > Select Board > Sipeed Maixduino

(3f.) Will only run the code for “helloworld.py”, although the LCD image orientation is misaligned, upside down; the picture is smaller than the LCD screen; but image has mostly clear resolution without any noticeable lag, latency, delay.

(3g.) Receiving “ImportError:  cannot import name MIC_ARRAY”
when attempting to run the code for:  

ROUTINE Sound Source Localization

https://wiki.sipeed.com/soft/maixpy/en/develop_kit_board/module_microphone.html 

*******and*******also*******for*******

AirPocket Soundman’s Sound Localizer
https://github.com/airpocket-soundman/SoundVisualizer/blob/main/src/mic_array_en_ver204.py

(3h.) Pin assignments used for MAix AMIGO

LED_CLK_GPIOHS28…….23

LED_DAT_GPIOHS27…….20

DAT 0…………….......……12

DAT 1……………..…..…….…9

DAT 2………………..…...……31

DAT 3………….……..……...7

WS……………..…....…..……19

BCK_SCK……...…..…………17

(4.) Would you list the correct pin assignments for the:

(4a.) MAix Amigo #2970 in the M1n position

(4b.) MAix Amigo #2970 in the STM position

(5.) Thanks.

(6.) Toodles!