sipeed/MaixPy-v1 #370: board_info problems

droneconnect, 2020-12-29:
Just uploaded the latest 6.2 firmware version and having issues with the board_info class.
A simple upload into boot.py such as:

from board import board_info
from fpioa_manager import *
board_test = board()
board_test.LED_G

produces a name 'board' isn't defined

also from a terminal I get : 
>>> from board import board_info
>>> from fpioa_manager import fm
>>> fm.
__class__       __module__      __name__        __qualname__
help            register        fpioa
get_pin_by_function             unregister      str_function
get_gpio_used
>>> fm.
__class__       __module__      __name__        __qualname__
help            register        fpioa
get_pin_by_function             unregister      str_function
get_gpio_used
**>>> board_info.
__class__       __module__      __name__        __qualname__
all             get             set             load**

any idea?

junhuanchen, 2020-12-30:
see this https://github.com/sipeed/MaixPy_scripts/tree/master/board

droneconnect, 2020-12-30:
thanks for the suggestion. I created the config.json and after it worked correctly

closed 2020-12-30 by droneconnect