sipeed/MaixPy-v1 #372: process images from the PC via UART

giodegas, 2021-01-03:
Could it be possible to send a picture grabbed from the laptop webcam to the Maix Dock via the UART USB-C connection?

If yes, what is the maximum bitrate achievable? Only QVGA format?

Neutree, 2021-01-04:
according to the USB-TO-UART chip, abour 1MiB to 4.5MiB(like board MaixPy Go), the MaixPy IDE use 1MiB

giodegas, 2021-01-04:
Is there any example available of image data transfer HOST --> MaixPy Dock board with MicroPython?

Neutree, 2021-01-04:
no example yet, but you can refer to pyserial official doc for PC;
And on MaixPy just use UART module to read data, it is easy, see https://maixpy.sipeed.com/zh/modules/on_chip/uart.html ,  and use`img = image.Image(jpeg_bytes, from_bytes=True)` to load jpeg as a image object, just try it