sipeed/MaixPy-v1 #375: M5StickV's LCD display not working with any firmware version > 0.5

domonless, 2021-01-06:
My M5StickV works fine with firmware which version < 0.6, like maixpy_v0.5.0_125_gd4bdb25_m5stickv.bin.

But after I flash any firmware which version > 0.5, such as maixpy_v0.6_9_g88bfd0177_m5stickv.bin,  my M5StickV can not display anything, including the power on logo.

I have tried the online compile tool many times, the same problem here.

junhuanchen, 2021-01-06:
After upgrading, you need to format flash once.

It will be rebuilt main.py

domonless, 2021-01-06:
> After upgrading, you need to format flash once.
> 
> It will be rebuilt main.py

Thank you for your reply, but I can't get it work.

I don't know how to properly flash it once and rebuilt main.py after upgrading.

I have tried use kflash erase all and erase partly before/after upgrading, no luck here.

And if I flash the 0.5 verion it works fine after rebooting.

So could you give me more instruction?

junhuanchen, 2021-01-06:
M5stackv lcd display code 
https://github.com/sipeed/MaixPy/blob/d78d036a2a74854974319d58f34e571615ec7d91/projects/maixpy_m5stickv/builtin_py/_boot.py#L83-L106

closed 2021-01-11 by junhuanchen