sipeed/MaixPy-v1 #389: OV5640 Timeout error

koda87, 2021-01-24:
Hi,
I have a problem with OV5640 camera, when I try to use it whit this code, i receive a timeout error, what can i do to solve it?
Thankyou 

`import sensor
import image
import lcd
import time

lcd.init(freq=15000000)
sensor.reset(choice=0)
print(sensor)
sensor.set_pixformat(sensor.GRAYSCALE)
sensor.set_framesize(sensor.SVGA)
sensor.run(1)
img = sensor.snapshot()
lcd.display(img)`

junhuanchen, 2021-01-25:
Ensure that the latest firmware.

change sensor.reset(choice=0) > sensor.reset().

it will auto scan sensor for ov5640.

koda87, 2021-01-25:
I had already try without "choice=0", I'm able to scan the sensor, the resul is 
Find ov5640 sensor 
and 
Find ov sensor. 
Seems that the sensor.reset find two sensors.
I tryed to insert choice=0 only for test.

koda87, 2021-01-25:
I attach my IDE versione and a print screen of code and error.
Thankyou for any help.

[image]

[image]

koda87, 2021-01-25:
I tryed whit the firmware versione 0.6.2_15 and now the software work fine (maixpy_v0.6.2_15_g0118a9a77 .bin).
How can I use the autofocus of this camera ?
How can I use VGA or SXGA resolution  ? I receive everything the Out Of Memory error.

[image]