sipeed/MaixPy-v1 #391: MaixPy compile failed.

willendzw, 2021-01-28:
Hi

I used latest code of MaxiPy to build. But it failed, error refer to below:

Scanning dependencies of target gen_collected_h
usage: /home/willen/Documents/MaixPy/components/micropython/core/py/makeqstrdefs.py command mode input_filename output_dir output_file
micropython/CMakeFiles/gen_collected_h.dir/build.make:57: recipe for target 'micropython/CMakeFiles/gen_collected_h' failed
make[2]: *** [micropython/CMakeFiles/gen_collected_h] Error 2
CMakeFiles/Makefile2:651: recipe for target 'micropython/CMakeFiles/gen_collected_h.dir/all' failed
make[1]: *** [micropython/CMakeFiles/gen_collected_h.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

Anyone can help or meet the same issue?

willendzw, 2021-01-29:
It can be closed. it is due to missing some codes.

closed 2021-01-29 by willendzw

sueng66, 2021-02-24:
what codes ? I have similar problem