sipeed/MaixPy-v1 #394: Download the MaixPy IDE.. impossible

fblgit, 2021-02-01:
Please... get that IDE somewhere, its impossible to download any file from your website, its constantly failing the download.. i wonder if u run the server on a 56k modem or what is the reason of not mirroring the IDE so people can download it...

Zepan, 2021-02-12:
you can download form the net disk, there is notice on the top of download website

fblgit, 2021-02-23:
have u consider to put it in some more places ? mostly coz in ur guide you state:
http://dl.sipeed.com/MAIX/MaixPy/ide/
But then inside there .. no mega/baidu drive.
then u have to go back to home and then u will see it.... "very obvious"

haomen, 2021-12-23:
+100 on this, why not update the download link directly to your Baidu drive url? It is very annoying trying to download 100 times but never succeed.  Or maybe consider put the release binary on github? At lease please consider put the notification (download large file from baidu drive) on every download page, this is just updating your download page HTML template.

12glace, 2022-01-27:
Where can I download MaixPy IDE? the download link doesn't work

12glace, 2022-01-27:
After an hour of searching, I finaly found a download link! here is the link for anyone that had trouble downloading it
: https://mega.nz/folder/5AAiSBwB#0jbOgqcrIMJVU9FvCzqb0A/folder/JQwkUBaZ

ZRFBelong, 2023-02-17:
Me neither !MaixPyIDE fail to connect to board,cause i have flash an unkown file called "demo.kfpkg_" ,andother operation with Visual Studio built-in arduino , i have no idea how to recovery to factory status