sipeed/MaixPy-v1 #395: How to use the mic array and the on-board mic simultaneously?

kevintsq, 2021-02-04:
I'd like to ask how to use the mic array and the on-board mic simultaneously? Since it seems impossible for me to get audio data from the mic array directly via MaixPy API, I tried to get audio data from the on-board mic while using the mic array to get sound direction. But it doesn't work. I guess there may be some I2S interface conflict between the on-board mic and the mic array. But it is said that the CPU supports 8 mics and 3 I2S devices, and the mic array contains 7 mics serving as 1 device. Could anyone helps me out? Thanks a lot~

RexzZzY, 2021-05-02:
You could try setting on board mic to another I2S device

For I2S device 0 (I2S0)

<< Pin Assignments may be different for your board and configurations>>

`fpioa = FPIOA()`
`fpioa.set_function(21, fpioa.I2S0_IN_D0)`  # 21/24/15/11(Single MIC) works
`fpioa.set_function(24, fpioa.I2S0_IN_D1)`
`fpioa.set_function(15, fpioa.I2S0_IN_D2)`
`fpioa.set_function(11, fpioa.I2S0_IN_D3)`
`fpioa.set_function(22, fpioa.I2S0_WS)`
`fpioa.set_function(23, fpioa.I2S0_SCLK)`

For I2S device 1 (I2S1)
`fm.register(20,fm.fpioa.I2S1_IN_D0, force=True)`
`fm.register(19,fm.fpioa.I2S1_WS, force=True)`    # 19 on Go Board and Bit(new version)
`fm.register(18,fm.fpioa.I2S1_SCLK, force=True)`  # 18 on Go Board and Bit(new version)`

Find more FPIOA  assignments on 
[maixpy FPIOA docs](https://wiki.sipeed.com/soft/maixpy/en/api_reference/Maix/fpioa.html)

petersmit, 2021-11-29:
Hi Kevin,

Did you get it working the mic array and the on-board mic. I have the same problem, I can not get it working. Can you share the code please?

Peter