sipeed/MaixPy-v1 #399: Maix Cube Audio

diazGT94, 2021-02-17:
Hello guys,

I’ve a Maixcube with the following firmware: maixpy_v0.6.2_27_g4d8d4fbf0_openmv_kmodel_v4_with_ide_support

I’ve tried the following routines, for playing the same sound described in the routines. https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/audio/play_wav.py
https://maixpy.sipeed.com/maixpy/en/course/media/audio.html?highlight=Audio

When I print the status of player variable it returns me 1 as value which in theory indicates that the audio is playing. However I can’t hear anything in my board. I also tried to set different volume values, so far nothing. Any suggestions?

junhuanchen, 2021-02-20:
see this https://github.com/sipeed/MaixUI/blob/master/driver/es8374.py

junhuanchen, 2021-02-24:
update https://maixpy.sipeed.com/maixpy/en/course/media/audio.html?highlight=Audio about es8374

closed 2021-03-08 by junhuanchen