sipeed/MaixPy-v1 #404: unable to acquire I2S on multiple channel

capgi, 2021-03-17:
on Maixduino, using record_wav.py modified to use sipeed mic array (pin assignement, configure I2S chanel 0to 3, received data seems to be ramdomized bytes

RexzZzY, 2021-05-02:
These are the configuration that I am using for my project

#Set FPIO for maixduino
`fpioa = FPIOA()`
`fpioa.set_function(21, fpioa.I2S0_IN_D0)`  # 21/24/15/11(Single MIC) works
`fpioa.set_function(24, fpioa.I2S0_IN_D1)`
`fpioa.set_function(15, fpioa.I2S0_IN_D2)`
`#fpioa.set_function(11, fpioa.I2S0_IN_D3)`
`fpioa.set_function(22, fpioa.I2S0_WS)`
`fpioa.set_function(23, fpioa.I2S0_SCLK)`

`rx0 = I2S(I2S.DEVICE_0)`
`rx0.channel_config(rx0.CHANNEL_0, rx0.RECEIVER, align_mode=I2S.STANDARD_MODE)`
`rx0.set_sample_rate(sample_rate)`

`rx1 = I2S(I2S.DEVICE_0)`
`rx1.channel_config(rx1.CHANNEL_1, rx1.RECEIVER, align_mode=I2S.STANDARD_MODE)`
`rx1.set_sample_rate(sample_rate)`

`rx2 = I2S(I2S.DEVICE_0)`
`rx2.channel_config(rx2.CHANNEL_2, rx2.RECEIVER, align_mode=I2S.STANDARD_MODE)`
`rx2.set_sample_rate(sample_rate)`

You should connect physical pins on board by referring the maixduino datasheet

jace, 2025-02-28:
@RexzZzY Are you able to record on all channels simultaneously? Calling `I2S.record()` uses DMAC_CHANNEL3 so won't they overlap? Code:

https://github.com/sipeed/MaixPy-v1/blob/d8901fd2272f000226f1c1037c1eb7c412b88e66/components/micropython/port/src/Maix/Maix_i2s.c#L287-L289

RexzZzY, 2025-02-28:
Unfortunately, this didn't work as expected @jace.  There was a ovelap