sipeed/MaixPy-v1 #406: Cannot use ampy or uPyLoader

paixaop, 2021-03-23:
**Describe the bug**
Cannot use ampy or uPyLoader with M5StickV using firmwares `maixpy_v0.6.2_37_g76f5aaaa4_m5stickv.bin` and `M5StickV_Firmware_v5.1.2.kfpkg`

**To Reproduce**
Steps to reproduce the behavior:
1. run `ampy --port /dev/cu.usbserial-C552BCD192 ls /`
2. See error

**Expected behavior**
Expected the list of files

**Actual behaviour**
```
b'[MaixPy] init end\r\n\r\n __  __              _____  __   __  _____   __     __\r\n|  \\/  |     /\\     |_   _| \\ \\ / / |  __ \\  \\ \\   / /\r\n| \\  / |    /  \\      | |    \\ V /  | |__) |  \\ \\_/ /\r\n| |\\/| |   / /\\ \\     | |     > <   |  ___/    \\   /\r\n| |  | |  / ____ \\   _| |_   / . \\  | |         | |\r\n|_|  |_| /_/    \\_\\ |_____| /_/ \\_\\ |_|         |_|\r\n\r\nM5StickV by M5Stack : https://m5stack.com/\r\nM5StickV Wiki       : https://docs.m5stack.com\r\nCo-op by Sipeed     : https://www.sipeed.com\r\n\r\n[MAIXPY]: result = 0\r\n[MAIXPY]: numchannels = 1\r\n[MAIXPY]: samplerate = 44100\r\n[MAIXPY]: byterate = 88200\r\n[MAIXPY]: blockalign = 2\r\n[MAIXPY]: bitspersample = 16\r\n[MAIXPY]: datasize = 158760\r\ninit i2c2\r\r\n[MAIXPY]: find ov7740\r\n'
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 8, in <module>
    sys.exit(cli())
  File "/Users/pedro/Library/Python/3.7/lib/python/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/pedro/Library/Python/3.7/lib/python/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/pedro/Library/Python/3.7/lib/python/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/pedro/Library/Python/3.7/lib/python/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/pedro/Library/Python/3.7/lib/python/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/3.7/site-packages/ampy/cli.py", line 194, in ls
    for f in board_files.ls(directory, long_format=long_format, recursive=recursive):
  File "/Library/Python/3.7/site-packages/ampy/files.py", line 162, in ls
    self._pyboard.enter_raw_repl()
  File "/Library/Python/3.7/site-packages/ampy/pyboard.py", line 192, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```

**Please complete the following information**
 - IDE version: 0.2.5
 - Firmware version: maixpy_v0.6.2_37_g76f5aaaa4_m5stickv.bin
 - Board: M5StickV
 - OS: macOS 11.2.3 Bigsur

junhuanchen, 2021-04-02:
try this https://github.com/junhuanchen/mpfshell-lite

zhengyangliu, 2021-12-27:
Same issue
Board: Maix Go
Firmware: maixpy_v0.6.2_72_g22a8555b5_with_lvgl.bin

uPyLoader is work fine for me, but none of cli tools is work.  mpfshell-lite ampy  or rshell.