sipeed/MaixPy-v1 #408: MaixAmigo - Front vs rear camera

PJPAC, 2021-03-30:
Hi,
I'm a newb with Sipeep and I'm trying to get a project running on MaixAmigo. 

MaixAmigo presents 2 cameras one on the front and the second on the rear of the device. However, all the camera related demos I found are using the front camera only.

See for instance demo_video_record.py

I cannot see how to select the rear camera as for instance there is no GC0328 / OV7740 label to set in the commands which would then allow using a specific camera. 
Could you please clarify how to select the camera rear and/or to suggest relevant documents ?

 - IDE version: 0.2.5
 - Firmware version: 0.6.2
 - Board: Maix Amigo
 - OS: Windows

junhuanchen, 2021-04-02:
see this https://github.com/sipeed/MaixUI/blob/master/test/test_amigo_sensor.py

closed 2021-04-02 by junhuanchen

PJPAC, 2021-04-14:
Thanks. 
It seems all down to sensor.reset(choice=1) vs sensor.reset(choice=2) instead of using sensor.reset(), and it works smoothly... 👍