sipeed/MaixPy-v1 #409: MaixAmigo - Cameras & TFT display resolution mismatch and limitation

PJPAC, 2021-03-30:
Hi, 
I'm a newb with Sipeep and I'm trying to get a project running on MaixAmigo. 

MaixAmigo presents 
-	1 camera on the front: GC0328 [VGA, Resolution 648 (H) x 488 (V) : 30 fps  -  www.gophotonics.com/products/cmos-image-sensors/galaxycore-microelectronics/21-117-gc0328]
-	1 camera at the rear: OV7740 [VGA (640x480): 30 fps ; QVGA (320 x 240): 60 fps - www.ovt.com/sensors/OV7740]
-	1 TFT Capacitive Touch Screen, 3.5'', 320x480

Demos such as demo_video_record.py are using 
    sensor.set_framesize(sensor.QVGA) 
corresponding to 320x240 which leaves a dark areas all around a central image on the LCD, and underuses the cameras which are 640x480. I was wondering about using VGA instead of QVGA even if the whole image does not show up on the 3.5” screen which is not really an issue especially when we use MaixPy IDE which itself includes a display. 
However, if we set 
    sensor.VGA
then after a few seconds we get the following error message 
    MemoryError: Out of Memory! Please reduce the resolution of the image you are running this algorithm on to bypass this issue!
with MaixPy IDE stopping at the following lines
    img_len = v.record(img)
If we attempt the same in demo_camera.py then the error message is “OSError: [Errno 12] ENOMEM” with MaixPy IDE stopping at “sensor.set_framesize(sensor.VGA)”. 

I would have expected that setting the natural resolution of the camera would have led to a zoom of the video being displayed on the 3.5” screen and the proper size appearing in the MaixPy IDE display. In parallel, if the amount of information were too high in VGA mode, we could have reduced the number of fps but it does not seem to have much impact on the error messages we get. 

It is very frustrating to have 640x480 cameras and being limited to 320x240. Could you suggest ways to bypass these issues, what compromise could be made, as well as to suggest relevant documents ?

Thanks

* System
 - IDE version: 0.2.5
 - Firmware version: 0.6.2
 - Board: Maix Amigo
 - OS: Windows

junhuanchen, 2021-04-02:
usb VGA too slow, should resize to 640*480.

try this https://github.com/sipeed/MaixUI/blob/master/driver/camera.py

https://github.com/sipeed/MaixUI/blob/master/ui/ui_camera.py

PJPAC, 2021-04-08:
Thanks,
camera.py kind of works with a 640*480 image appearing in the MaixPy IDE. Intriguingly, only a section of this image appears on the top right side of the 3.5” TFT screen. Not a zoom but a fraction of the image with a grey area on the left and bottom of this image. It is not yet clear how to zoom or centre a fraction of the 640*480 image on the TFT display. 

For ui_camera.py I get "ImportError: no module named 'ui'" from the line "ui.ui_canvas import ui" but as I'd like to use K210 I'm unclear whether or not l should install drivers related to ui.