sipeed/MaixPy-v1 #410: MaixAmigo - acceleration sensor

PJPAC, 2021-03-30:
Hi,
I'm a newb with Sipeed and I'm trying to get a project running on MaixAmigo. 

MaixAmigo apparently presents an acceleration sensor (www.seeedstudio.com/Sipeed-Maix-Amigo-p-4689.html). However, I could find neither demos, nor documentations, nor command line/sensor ID I could try to start from. Furthermore, this acceleration sensor does not seem to be included in config_maix_amigo.py. 
Would someone point in some directions ?

Thanks

* System
 - IDE version: 0.2.5
 - Firmware version: 0.6.2
 - Board: Maix Amigo
 - OS: Windows

junhuanchen, 2021-04-02:
https://maixpy.sipeed.com/maixpy/en/develop_kit_board/maix_amigo.html

[image]

closed 2021-04-02 by junhuanchen

PJPAC, 2021-04-14:
Thanks. I could start playing around. :)
I have noticed that even when the Amigo is not moving, small accelerations are measured along the 3 directions XYZ so that I'll see to set thresholds to trigger anything in a more reliable way. The gravity is always 10% off, from whichever value is being used in the _STANDARD_GRAVITY parameter, but it might just be due to the sensor it-self (?).
So far I've been testing with the Amigo connected to my computer and interfaced through MaixPy IDE. While editing the program I have had a few error messages which I could not match with the program writing it-self, but which disappeared after disconnecting and reconnecting the IDE and Amigo devices, as if it was needed for the program to be compiled... 
The next step is to run the program.py directly from the Amigo it-self so that I don't have to carry the laptop, while walking or running. However when I turn on the Amigo, I don't see how to pass the welcome page with the red background, and I don't see yet how to run a specific program.py. It is probably obvious to a more experienced user (I'm still a newb), but if someone could point at some directions it would be greatly appreciated.
Thanks.