sipeed/MaixPy-v1 #414: MaixPy project build error for TensorFlow Lite Micro Hello world example

marconi1964, 2021-04-21:
**Describe the bug**
I tried to build MaixPy with TensorFlow Lite Micro hello world example, but fail in finding *.h in MaixPy build. I am not familiar with cmake build, does not know how to add "include" directory in source directory CMakeLists.txt.  The (partial of) CMakeLists.txt in source directory is listed below.

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Build TensorFlow Lite Micro (TFLM) source code, and copy source code from TFLM to maixpy directory.
```
# download MaixPy
$ git clone https://github.com/sipeed/maixpy
# Create a new directory for TFLM hello world
$ mkdir -p maixpy/projects/TFLM_hello_world
# Copy MaixPy hello world template directory to TLFM hello world
$ cp -r maixpy/projects/hello_world/ maixpy/projects/TLFM_hello_world/
# download TensorFlow
$ git clone https://github.com/tensorflow/tensorflow.git
$ cd tensorflow
# TensorFlow Lite Micro build source files structure
$ make -f tensorflow/lite/micro/tools/make/Makefile TARGET=riscv32_mcu generate_hello_world_make_project
$ cd tensorflow/lite/micro/tools/make/gen/riscv32_mcu_riscv32_mcu_micro/prj/hello_world/make
$ ls -l
total 40
-rw-r--r-- 1 ubuntu ubuntu 15503 Apr 20 07:43 LICENSE
-rw-r--r-- 1 ubuntu ubuntu 11291 Apr 20 07:43 Makefile
-rw-r--r-- 1 ubuntu ubuntu   810 Apr 20 07:43 README_MAKE.md
drwxr-xr-x 3 ubuntu ubuntu  4096 Apr 20 07:43 tensorflow
drwxr-xr-x 6 ubuntu ubuntu  4096 Apr 20 07:43 third_party

$ cp -r ./  ~/maixpy/projects/TFLM_hello_world
```
2. Modify source directory CMakeLists.txt "Add source files", based on TFLM Makefile content (SRCS:= \.....)

```
$ cd maixpy/TFLM_hello_world/main/
$ cat CMakeLists.txt
############### Add include ###################
# list(APPEND ADD_INCLUDE "include"
#    )
# list(APPEND ADD_PRIVATE_INCLUDE "")
###############################################

############ Add source files #################
list(APPEND ADD_SRCS 
"tensorflow/lite/micro/simple_memory_allocator.cc"
"tensorflow/lite/micro/simple_memory_allocator.h"
"tensorflow/lite/micro/micro_error_reporter.cc"
"tensorflow/lite/micro/all_ops_resolver.cc"
"tensorflow/lite/micro/memory_helpers.cc"
....
"tensorflow/lite/micro/examples/hello_world/main.cc" 
"tensorflow/lite/micro/examples/hello_world/main_functions.cc" 
"tensorflow/lite/micro/examples/hello_world/model.cc" 
"tensorflow/lite/micro/examples/hello_world/output_handler.cc" 
"tensorflow/lite/micro/examples/hello_world/constants.cc" 
"tensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/libwrap/misc/write_hex.c" 
"tensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/libwrap/stdlib/malloc.c" 
"tensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env/start.S" 
"tensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env/entry.S" 
"tensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env/freedom-e300-hifive1/init.c"
"tensorflow/lite/micro/arduino/abi.cc"
    )
# aux_source_directory(src ADD_SRCS)
# list(REMOVE_ITEM COMPONENT_SRCS "src/test2.c")
###############################################

###### Add required/dependent components ######
list(APPEND ADD_REQUIREMENTS kendryte_sdk)
###############################################

############ Add static libs ##################
# list(APPEND ADD_STATIC_LIB "lib/libtest.a")
###############################################

register_component()
```

3. MaixPy build and error occurs

```
$ cd ~/maixpy/projects/TFLM_hello_world
$ python3 project.py build
.....
/home/ubuntu/maixpy/projects/amigo_tflm/main/tensorflow/lite/micro/micro_error_reporter.cc:16:10: fatal error: tensorflow/lite/micro/micro_error_reporter.h: No such file or directory
 #include "tensorflow/lite/micro/micro_error_reporter.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/ubuntu/maixpy/projects/amigo_tflm/main/tensorflow/lite/micro/simple_memory_allocator.cc:16:10: fatal error: tensorflow/lite/micro/simple_memory_allocator.h: No such file or directory
 #include "tensorflow/lite/micro/simple_memory_allocator.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

**Please complete the following information**
 - IDE version: 
 - Firmware version:
 - Board: Maix Amigo
 - OS: Linux Ubuntu 18.04

zhongpanwu, 2022-06-17:
Hi, have you solved this problem , could you give me some clues, thank you

Neutree, 2022-06-18:
uset `python project.py rebuild --verbose` to see what dependence or something wrong