sipeed/MaixPy-v1 #422: Error :ValueError:[MAIXPY] kpu:load error:2006 , ERR_No_MEM: memory not enough

sbabums, 2021-06-04:
**Describe the bug**
When I try to execute the driver program from the MaixPy IDE i get "ValueError:[MAIXPY] kpu:load error:2006 , ERR_No_MEM: memory not enough".

From kpu.memtest(), the following are the outputs:
###free gc heap memory:494 KB
###free sys heap memory:2052 KB

The model size is 3336 KB. 

From the above, i could understand that the size of the model is greater than the free sys heap memory, therefore mem error is encountered. However I want to find a way to resolve this issue. If anybody have faced this issue and if you have tried to solve it , please share the steps. Thanks!!!

**To Reproduce**
Steps to reproduce the behavior:
1. Using Kflash GUI loaded the model and firmware
2. Using MaixPy IDE executed the driver program after connecting the Maix Dock to the serial port

**Expected behavior**
The driver program should be able to load the model onto flash memory and detect the objects.

**Actual behaviour**
When I try to execute the driver program from the MaixPy IDE i get "ValueError:[MAIXPY] kpu:load error:2006 , ERR_No_MEM: memory not enough".

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following information**
 - IDE version: [0.2.5]
 - Firmware version: [0.5.0]
 - Board: [Maix  Dock]
 - OS: [Windows]
 - 
[image]

junhuanchen, 2021-06-11:
https://github.com/sipeed/MaixPy/issues?q=memory+not+enough

closed 2021-06-11 by junhuanchen

No1possible, 2023-06-11:
Replace with a smaller firmware