sipeed/MaixPy-v1 #423: load error:2015 when trying to load model with kpu

Balocre, 2021-06-26:
I've flashed a model compiled from onnx with nncase 1.0 CI but I get this error when trying `kpu.load(0x300000)` :

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: [MAIXPY]kpu: load error:2015, ERR_UNKNOWN
```
Any idea what could be causing it? The model I compiled is the one described here : https://github.com/kendryte/nncase/issues/248 .
I've tried several variations whith deifferent op, but nothing worked.

Balocre, 2021-06-26:
I've also tried with and without `--quant-type` option too, just in case, but same error

Balocre, 2021-06-26:
I just tried with `kpu.load_flash(0x00300000, 0, 0, 0x4000)` and I get :

```
SPI freq 16384 Hz
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: [MAIXPY]kpu: load_flash error:2002, ERR_KMODEL_VERSION: only support kmodel V3/V4 now
```

Balocre, 2021-06-26:
The version of MaixPy I use is `maixpy_v0.6.2_54_g897214100_minimum_with_kmodel_v4_support.bin` so V4 should be supported

KuoEuran, 2021-07-12:
@Balocre Have you solved the problem yet?

Balocre, 2021-07-13:
> @Balocre Have you solved the problem yet?

The issue is that I compiled my model using nncase CI version which outputs kmodel version 5 files. So I am working with kendryte SDK for now, until maixpy supports V5 (which might be not so soon imo)..

KuoEuran, 2021-07-27:
@Balocre  Hi, so the answer is use the kendryte SDK, and your model can run on the k210 now? tks

Balocre, 2021-07-27:
> @Balocre Hi, so the answer is use the kendryte SDK, and your model can run on the k210 now? tks

I have been able to load the model without errors and get it running without crashing, but I am unable to confirm it was working properly as the MaixDuino camera model I have (gc0328) is not supported natively by the SDK, I have been trying to port the driver from MaixPy, but I couldn't get the camera to work and  i have no idea why.

ov2640 (the former model of camera included in the Sipeed MaixDuino kit) worked out of the box with the drivers provided in the SDK, but I stopped working with the MaixDuino for now because the developers of MaixPy don't really seem to care about the issues, and their code is an absolute nightmare to read.

KuoEuran, 2021-07-27:
@Balocre Thank you for your replying. Seems like the own customed model can use on C sdk, but the other firmware is not support like camera. I use Maxiduino too. So what's the next board you use to run your program or you give up?

Balocre, 2021-07-27:
> @Balocre Thank you for your replying. Seems like the own customed model can use on C sdk, but the other firmware is not support like camera. I use Maxiduino too. So what's the next board you use to run your program or you give up?

I will be developing on Nvidia Jetson, I will check back later if by miracle MaixPy has been updated